react-native-camera/react-native-camera #118: no callback when recording video

klausbreyer, 2015-11-06:
I want to reopen this, because I have no callback on video :( 

```          <Camera
        ref="cam"
        style={styles.container}
        // onBarCodeRead={this._onBarCodeRead}
        type={this.state.cameraType}
        captureMode={Camera.constants.CaptureMode.video}
        captureTarget={Camera.constants.CaptureTarget.disk}
      >
    
       <View style={styles.controls}>
        <TouchableOpacity  onPressIn={this.record.bind(this)}  onPressOut={this.pause.bind(this)} style={styles.controlBtn}>
          <Text>Record</Text>
        </TouchableOpacity>
        </View>
        
      </Camera>

 ````

```
  record() {
      console.log('record');
      this.setState({
          recording: true
      })
      this.refs.cam.capture(function(err, data) {
          console.log(err, data);
      });
  }

  pause() {
          console.log('pause');
          this.refs.cam.stopCapture()
  }
```

with Camera.constants.CaptureMode.video I get an callback. What could possibly be wrong here?

lwansbrough, 2015-11-06:
Just to verify this isn't a JS issue, can you do `this.refs.cam.capture({}, function (err, data) {...` ?

klausbreyer, 2015-11-09:
Hey, 

thanks for your reply. I figured out, that the video is captures on the real device (including callback) but not on the simulator. 

is this an general issue? here is my full code:

'use strict';
```
var React = require('react-native');
var {
  AppRegistry,
  View,
  Animated,
  StatusBarIOS,
  CameraRoll,
  TouchableOpacity,
  TouchableHighlight,
  TouchableWithoutFeedback,
  Text,
  StyleSheet,
  NavigatorIOS
} = React;

// var screen    = require('Dimensions').get('window');
var Camera = require('react-native-camera');

// var CONFIG = require('./CONFIG');
// var Preview = require('./Preview');

class Cam extends React.Component {
  constructor(props) {
    // // console.log("konstruktor rekord")
    super(props);
    this.state = {
       cameraType: Camera.constants.Type.back,
               
    }

  }

  render() {
 
    return (
 <Camera
        ref="cam"
        style={styles.container}
        onBarCodeRead={this._onBarCodeRead}
        type={this.state.cameraType}
      >
        <TouchableHighlight onPressIn={this._record.bind(this)} onPressOut={this._done.bind(this)} >
          <Text style={{color: 'white'}} >Take Picture</Text>
        </TouchableHighlight>
      </Camera>
    );
  }
   _onBarCodeRead(e) {
    console.log(e);
  }
  _switchCamera() {
    var state = this.state;
    state.cameraType = state.cameraType === Camera.constants.Type.back
      ? Camera.constants.Type.front : Camera.constants.Type.back;
    this.setState(state);
  }
  _record() {
    console.log('record');
   this.refs.cam.capture({audio: true, mode: Camera.constants.CaptureMode.video, target: Camera.constants.CaptureTarget.cameraRoll}, function(err, data) {
            console.log(err, data);
        });

  }
  _done() {
       console.log('pause');
          this.refs.cam.stopCapture()

  }

}

var styles = StyleSheet.create({
  container: {
    flex: 1,
    justifyContent: 'center',
    alignItems: 'center',
    backgroundColor: 'transparent',
  },
  welcome: {
    fontSize: 20,
    textAlign: 'center',
    margin: 10,
  },
  instructions: {
    textAlign: 'center',
    color: '#333333',
  },
});

Cam.external = true;
module.exports = Cam;

```

ms88privat, 2015-11-10:
@klausbreyer what I know is, that you can never use any camera in the simulator

klausbreyer, 2015-11-10:
omg! 
This is something we should wirte in the readme :) I thought it just keeps white. which would be ok for testing reasons.. 

thank you!

lwansbrough, 2015-11-10:
Yeah, very important to mention your testing parameters!

closed 2015-11-10 by lwansbrough