react-native-camera/react-native-camera #119: Taking cropped photo

vinu123, 2015-11-09:
I want my camera view to be of square size i.e. half of my iPhone's screen. I tried using 'Aspect' => 'fit' however that is only for the camera view and not the actual picture. How can I restrict the picture to be my camera view's size?

lwansbrough, 2015-11-10:
The camera doesn't support this and generally I'd like to keep it out of the camera. Processing should be left to a post-processing module.

lwansbrough, 2015-11-13:
Take a look at `ImageEditingManager` that ships with React Native.

closed 2015-11-13 by lwansbrough

vinu123, 2015-11-24:
Hey @lwansbrough, How can I use ImageEditingManager?

achang28, 2015-11-24:
@vinu123 :  https://github.com/facebook/react-native/blob/master/Examples/UIExplorer/ImageEditingExample.js

vinu123, 2015-11-24:
What I don't understand is that do I use <SquareImageCropper /> because this example exports that but when I do use it, I get an error. I just don't know how to use it. God, I feel so dumb :(

jhack32, 2016-11-18:
If you'd like the camera to take the same image size as an iPhone camera. Check the link below. You'd need to adjust your preview to be the same as the iPhone camera.

https://github.com/cortlandtlabs/react-native-camera/pull/1