react-native-camera/react-native-camera #123: torchMode not turning on light

brancusi, 2015-11-18:
Hi,

I would like to enable torchMode during barcode scanning. I am setting the following:

```javascript
<Camera
   onBarCodeRead={::this._onBarCodeRead}
   type={Camera.constants.Type.back}
   torchMode={Camera.constants.TorchMode.on}
/>
```

For some reason this is not starting the torch though. Everything else works just fine, camera view and scanning.

What's strange is that when I navigate away then back to the scene, I see a flicker of the torch but it shuts off.

Device info: iphone 6 plus
OS: ios 9.1

Any ideas?

lwansbrough, 2015-11-26:
Interesting, not sure about this. I'll look into this when I get a chance. If you're familiar with Obj-C I might recommend you dive into it a bit yourself if you're willing - might be faster than waiting for me. I'd definitely accept a PR if you discover the issue.

cava23, 2015-12-19:
So far I've traced this down to the setType method in RCTCamera.m. Torch mode is on until initializeCaptureSessionInput:AVMediaTypeVideo is called in setType. At this point, the currently configured device input is replaced, disabling torch mode. If I comment out the call to initializeCaptureSessionInput, torch mode stays on correctly. Commenting that out probably breaks something else, but that's just as far as I've gotten. @seanj

cava23, 2015-12-19:
Here ya go, wrap setType with:
```
if (self.manager.presetCamera != type) {
   // change the camera type
}
```
I'll make a PR and see what you think.

rajatmehndiratta, 2016-02-11:
Running into the same issue; was @cava23's PR made? I can't find it when I search + don't see the change in RCTCamera.m.

cava23, 2016-02-11:
I couldn't quite get it right, so I never made a PR. We ended up initializing the camera on componentDidMount, after the props were set. You can see our fork here: https://github.com/Tredsite/react-native-camera

rdhaliwal, 2016-02-15:
FWIW @brancusi I had the same issue where I tried to initialise the Camera with TorchMode.on, but when I changed it so it was set with a variable and it was initially set to off, then toggled to on, it started to work for me. Maybe it'll help? ¯\_(ツ)_/¯

```
<Camera torchMode={this.props.flash ? Camera.constants.TorchMode.on : Camera.constants.TorchMode.off}
                ref={(cam) => { this.camera = cam; }} />
```

Obooman, 2016-05-06:
I find something strange here:When I set torchMode to `Camera.constants.FlashMode.on`(yea no spelling mistake),the torch turned on for a while( about 1 seconds ).
When I set trochMode to `Camera.constants.TorchMode.on`,it doesn't work at all.

I console.log the `Camera` object and find that both  `Camera.constants.FlashMode.on` and `Camera.constants.TorchMode.on` are the number `1`.That's weired.

*Device Info:iPod touch with iOS 9.3.1(13E238)*

code here
```javascript
<Camera 
	torchMode = {1}
	onBarCodeRead = { this.barcodeHandler }
/>
```

yutin1987, 2017-02-23:
when run this code `[self.session removeInput:self.videoCaptureDeviceInput];` https://github.com/lwansbrough/react-native-camera/blob/master/ios/RCTCameraManager.m#L520
will trigger off torch

dadbob, 2017-04-04:
In initializeCaptureSessionInput, add this: `        dispatch_async(self.sessionQueue, ^{
            NSInteger *torchMode = 1;
            AVCaptureDevice *device = [self.videoCaptureDeviceInput device];
            NSError *error = nil;
            
            if (![device hasTorch]) return;
            if (![device lockForConfiguration:&error]) {
                NSLog(@"%@", error);
                return;
            }
            [device setTorchMode: torchMode];
            [device unlockForConfiguration];
        });` after this: `      captureDevice = [self deviceWithMediaType:AVMediaTypeVideo preferringPosition:self.presetCamera];
`

mesbahsabur, 2017-05-01:
I found that the issue was caused by using a button which i positions 'absolute' over the camera component. Solved this by setting the camera as absolute and the buttons as regular views, like this:

Hope it helps!

```
render() {
    return (
      <View style={styles.container}>

        <View style={styles.cameraContainer}>
          <Camera
            torchMode={this.state.torch ? Camera.constants.FlashMode.on : Camera.constants.TorchMode.off}
            ref={(cam) => { this.camera = cam; }}
            // rest
          >
          </Camera>
        </View>

        <View style={styles.buttonContainer}>
          <TouchableOpacity onPress={() => this.onTorchButton()}>
            <View style={styles.button}>
              {this.state.torch ? <Icon color='white' type='material-community' name='flashlight-off' /> :
              <Icon type='material-community' name='flashlight' /> }
            </View>
          </TouchableOpacity>
        </View>

      </View>
    );
  }

const styles = StyleSheet.create({
  container: {
    flex: 1,
  },
  buttonContainer: {
    flex:1,
    padding:15,
    flexDirection: 'row',
    alignItems:'flex-end',
    justifyContent:'center'
  },
  cameraContainer: {
    position: 'absolute',
    width: width,
    height: height,
    top:0,
    left:0
  },
  button: {
    margin:20,
    flex:0,
    borderRadius:25,
    width:50,
    height:50,
    alignItems:'center',
    justifyContent:'center',
    backgroundColor:'red'
  },
});
```

samuela, 2018-01-04:
I also found that setting the `torchMode` to a constant `Camera.constants.TorchMode.on` doesn't work. Not sure why. My workaround looks something like

```javascript
export default class App extends Component {
  constructor () {
    super()
    this.state = {
      torchOn: false
    }
  }

  componentDidMount () {
    this.setState({torchOn: true})
  }

  render () {
    return (
      <View style={styles.container}>
        <Camera
          torchMode={this.state.torchOn ? Camera.constants.TorchMode.on : Camera.constants.TorchMode.off}
          // other stuff here...
        />
      </View>
    )
  }
}
```

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

samuela, 2018-01-12:
@ospfranco I'm still having this issue as of 0.12.0.

ospfranco, 2018-01-12:
ok, re-opening

Nauzer, 2018-02-07:
Same here on 0.13.0

sibelius, 2018-02-07:
can u try master with RNCamera?

enieber, 2018-08-30:
@sibelius  this issue can it is being closed?

closed 2018-08-30 by sibelius