react-native-camera/react-native-camera #125: Undefined is not an object(evaluating 'Camera.constants.Type')

andrzey, 2015-11-18:
Hi! I'm trying to create a simple cameraapp on Android with your react-native-camera module. But when I try to use it I get the error-message: "Undefined is not an object(evaluating 'Camera.constants.Type')". I have installed the module with the npm command and react-native-camera is present in node-modules in my react-native-app. Is there something I have forgotten to do? Would really appreciate if you could help me a little!

mikeljames, 2015-11-25:
Hi @andrzey Did you also get it working on iOS? I'm getting the same issue.

lwansbrough, 2015-11-26:
@mikeljames Should work for iOS.

mikeljames, 2015-11-26:
Yeah I Failed to follow the install instructions correctly. I'm new to this react-native stuff. I've got a few other issues, but +1 existing issues. :)

3210jr, 2015-11-30:
hey @andrzey what are the additional things that need to be done to android? I am having a hard time figuring it out. Any help will be greatly appreciated! Thanks.

alenave, 2016-05-02:
@andrzey what are the additional things to be done, to work Android

alenave, 2016-05-02:
hey @3210jr did you get the solution for the error?

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco