react-native-camera/react-native-camera #127: Why is the memory option of the captureTarget prop deprecated?

gorkemyurt, 2015-11-23:
Looks like the Camera.constants.CaptureTarget.memory property is deprecated, was there a reason for this? I am guessing sending the encoding data to the JS side was too expensive? 

Does that mean the default value should be the assets library url not the base 64 encoded string? 

That would make the information following the depreciation info outdated..
"This property allows you to specify the target output of the captured image data. By default the image binary is sent back as a base 64 encoded string."

lwansbrough, 2015-11-26:
@machard Is working on an improvement for memory targets, which will actually bring it out of deprecation! :+1: The necessary changes to React Native core will be coming in 0.16.

thomasjonas, 2015-12-21:
It would be greatly helpful for my project if this feature will be brought out of deprecation! Any news on this?

blargism, 2016-05-18:
I'd like to add additional reasons for why this option is useful.

**TL;DR:** I have to deal with HIPAA requirements, keeping images around past their immediate use can result in a violation of those requirements.  I need to know they will go away at app closure or crash.  In memory does this nicely.  If temp files also do this nicely please advise.

**More Detail**

This is a very important feature for my use case.  I have to deal with HIPAA constraints so storage of a photo, even in temporary files, makes things more difficult.  It would help us stay in compliance if we can easily discard an in memory image and even more helpful as in memory image data is discarded automatically on app closure or crash.

In the event of an application crash an in-memory image would just be discarded with the rest of the app's memory.  If we are working with temporary files, these files could potentially be orphaned.  While it is not likely, it is possible that a temp image saved outside of the app sandbox would be exposed outside our app. This is problematic and a potential HIPAA violation.

While we are doing all we can to ensure our app does not crash, stuff happens.  Sometimes you can't anticipate those issues.  Volatile memory based images mitigates our risk in most of these cases.

**Caveats**

Note: these caveats must be true on both Android and iOS.

If the implementation of temporary files is sufficient to ensure that on app closure all temp files are thrown out then this may be a moot issue.  I'm not familiar enough with these systems to know for sure.  Please advise if this is the case.

Also, if your implementation ensures that temporary files are stored in the app's sandbox this may also be acceptable.

Thanks for taking the time to look into this.

rt2zz, 2016-05-18:
I think there is enough community interest that we will likely un-deprecate the base64 option, although it will likely be renamed in the near future.  

@blargism please do due diligence on the code base. Based on what you mentioned above I think either `target="temp"` or `target="memory"` will work for you. Temp stores it in the temp files directory which to my knowledge is inside of the app's sandbox. Memory will generate a base64 sring and not save anything to disk.

blargism, 2016-05-18:
@rt2zz thank you for your quick response.

As you asked me to research the issue I thought I might provide my findings here so that people do not have to repeat my work.

iOS and Android cache systems are used in the instance of the "temp" file option.  On both platforms this plugin uses the sandboxed location for these files.  However, this still requires that those files be managed by the user.  The OS does clean these files up, but not in any reliable way.

For those with HIPAA concerns (such as myself), the "temp" option would require you to manage the deletion of the temporary files yourself.  However, those files are only accessible by the application in question.  These files are effectively closed off from the OS and other apps on both platforms.  Implementation of the "temp" option should still take into account the need to clean up files as on both iOS and Android those files are not deleted until the OS does file cleanup.  Mostly that file cleanup happens due to the OS detecting it needs more storage space.

There are several plugins that help with file storage, the top google result is https://github.com/johanneslumpe/react-native-fs

rt2zz, 2016-05-18:
@blargism awesome! thanks for checking into that

mlund01, 2017-01-03:
+1 any update on un-deprecating this feature?

dggriffin, 2017-02-24:
Any news here?

lsps9150414, 2017-05-10:
@blargism Thanks for sharing! I'm also dealing with HIPAA. Would you mind share more on how you've achieved that? Did you go with `temp` and manually clean up the files with `fs` or did you find other way to do this?

blargism, 2017-05-10:
It's been awhile, but I think I used the in memory option.  I just couldn't risk the pictures getting to someplace where google or apple would pick them up and auto-save them to the cloud.  That little feature from the phone manufacturers means a HIPAA breach, which is scary bad.

fxfactorial, 2017-08-03:
Also curious as to the reasoning behind deprecating it, but it seems like de facto it is still a viable option.

arnsa, 2017-08-03:
+1, also interested in using memory option, but not sure if it's okay using it, since it's deprecated?

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco