react-native-camera/react-native-camera #132: dyld: Symbol not found: _AVMetadataObjectTypeDataMatrixCode

zubru, 2015-12-04:
Hi,  
I upgraded to newest react-native and xcode. I tried run my previous app on device and I'm getting error:

```dyld: Symbol not found: _AVMetadataObjectTypeDataMatrixCode
  Referenced from: /var/mobile/Applications/E3166042-F289-4E33-B721-FFF31236AE1E/appname.app/appname
  Expected in: /System/Library/Frameworks/AVFoundation.framework/AVFoundation
 in /var/mobile/Applications/E3166042-F289-4E33-B721-FFF31236AE1E/appname.app/appname```

When I uninstall react-native-camera it starts fine. 
Any solutions to that problem?

lwansbrough, 2015-12-08:
A PR has been merged to address this. Not sure if I've created a new npm version for all the recent changes yet though.. Sorry about that, been busy.

lwansbrough, 2015-12-08:
Try installing from this master branch and see if that fixes the problem.

zubru, 2015-12-14:
Hi,
Tried your fix and I think it works. Unfortunately I have some issues with new react-native. But that error with your camera module is fixed.
Thanks a lot!

lwansbrough, 2015-12-14:
Great!

lwansbrough, 2015-12-14:
Actually I'll keep this open until a new version of the camera is released.

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco