react-native-camera/react-native-camera #133: Metadata usage unclear.

dangnelson, 2015-12-11:
I can't for the life of me figure out how to get metadata based on your docs. Do you think you could provide an example in the documentation?

dangnelson, 2015-12-11:
I'm assuming its something like:

     this.refs.cam.capture({ meta: true}, function(err, data) {
        console.log(data);
    }

But it doesn't seem to be logging anything but the image location.

dangnelson, 2015-12-11:
It seems to me that getting back an object with all the image data as default would be more useful as default behavior as the native camera and library do on iOS:

    { isVertical: true,
      width: 2448,
      uri: 'file:///var/mobile/Containers/Data/Application/DC903F36-04CA-4976-B799-5B2719F3B9AC/Documents/images/1B8F9C5F-CFA8-4C97-B533-52E0F981513E.jpg',
      height: 3264 }

dangnelson, 2015-12-11:
I suppose my request is in line with this one:
https://github.com/lwansbrough/react-native-camera/issues/117

I'm trying to crop to an image on capture, but I need the image width and height information in order to calculate the amount to crop using the Native Module.

lwansbrough, 2015-12-18:
Hey Dan, I agree. I think this is probably better. I want to make some time to get in some of the changes I want done and get a new version out - but I can't make any promises in the short term. You could use an intermediate step which gets the image metadata and then perform the crop after.

sscaff1, 2016-03-30:
@lwansbrough this seems very similar to #188.

ptomasroos, 2017-04-26:
Closing in favor of #188

closed 2017-04-26 by ptomasroos