react-native-camera/react-native-camera #134: Check permissions to access Camera

bpeters, 2015-12-13:
If a user denies permissions for the camera and then tries to access the camera the app will crash. Is there a way to check for camera permissions from the Camera Api? I can then request permission with an alert like the default.

lwansbrough, 2015-12-14:
Apparently I forgot to expose the `checkDeviceAuthorizationStatus` via the JS API. You can do what you're asking via:

```javascript
NativeModules.CameraManager.checkDeviceAuthorizationStatus((function(err, isAuthorized) {}
```

I'd accept a PR if you want to expose that method on the `Camera` reference like others have been in the `index.ios.js` file.

closed 2015-12-14 by lwansbrough

tonmanayo, 2018-08-13:
Hi, I'd like to know if there is any way of checking Camera permissions. I'd like to being up a modal before asking the users on IOS for permissions