react-native-camera/react-native-camera #135: Access to camera feed as stream

nickfujita, 2015-12-15:
Hello! First off, thanks for all the great work on this project, my team and I are extremely thankful that this is getting built out!

I was wondering if it would be at all possible to get access to the video stream from the camera. Currently it seems that for video, the only option would be to capture the stream to a static file, and providing a link to that file in storage (disk or camera roll).

For a second, the memory option looked promising, but it seems to be deprecated so I wan't able to test it, but I am guessing it provided the base64 encoded data only after the fact of the video being recorded, and not as a stream.

What I am really looking to play around with is attempting to make a periscope or meerkat clone as a learning experiement as we explore the features and limitations of the current scope of React Native, but can't seem to get access to a camera feed. So far we have considered this module or screcorder, but it seems that both are more focused on static data.

Any direction, assistance or advice would be greatly appreciated! Thanks in advance!

lwansbrough, 2015-12-18:
So, a couple of things:

- I plan to un-deprecate the memory option because of functionality made available in core React Native (see this PR: https://github.com/lwansbrough/react-native-camera/pull/100). The memory option only provides the ability to receive a reference to a still image though.

- For any option you choose, you do not want to stream data into the JS side, because that will lock up the bridge which is a non-starter for React Native. If you wanted to use this module, you'd need a way to control the stream so you can get the reference to it in another native module. You'd probably have to build the native module for this (unless one exists - but I doubt it) and that native module would read the native camera stream from this module and do the streaming to your server itself. There may be a way to store generic references and restore them in different locations; I've done this before in other custom native modules, but then it becomes a question of how to do garbage collection.

Unless something has changed recently, there isn't good native module interop yet in React Native. It's something we desperately need IMO. But that doesn't mean we can't do it now, it's just harder.

jacktuck, 2016-11-22:
Would love this.

sashakid, 2017-02-16:
Now I am just working on making access to camera feed as stream. @lwansbrough what did you say about providing stream data to JS side? It's not possible or it doesn't make sense? Though, if I want to have the stream on JS side, what type should it be? Base64 string?

henrikra, 2017-03-24:
Does this help at all? http://stackoverflow.com/questions/12242513/how-to-get-real-time-video-stream-from-iphone-camera-and-send-it-to-server

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

Ouwen, 2018-01-26:
@lwansbrough @sashakid 

I am also curious on how to enable data streams through react native. Seems like it would be a valuable feature. You guys probably have better background on react native than I do, but I was wondering why naively we couldn't spawn off a worker to stream data. This way the event loop react native is using is not blocked. (https://www.npmjs.com/package/react-native-workers)

Let me know if there is something I might not be thinking of here.

MissAnichka, 2018-01-30:
I would love to be able to capture the live camera feed! Has anyone figured this out yet? Or a workaround?

sdhnshu, 2018-04-04:
Need this

daaain, 2018-07-31:
Well there's this post which explains a DIY method with patching the library: https://medium.com/react-native-training/build-youtube-alike-livestreams-with-react-native-8dde24adf543

The author said he'll submit a PR, but it was only a few days ago...

sibelius, 2018-07-31:
feel free to send a PR based on that approach

AngelVenchev, 2018-12-12:
Hey I am working on this feature since it's a very important part of our app. I've done the dynamic file reading (while the video is still being recorded). Still the issue is that the bridge is being locked up as. @lwansbrough said in the beginning of this thread. 
I am writing here to ask for any directions or documentation on how to implement this with a separate react-native module. I am determined to finish this since we need it for our app and I have been given enough time to implement this. So any help with the separate module streaming the data to the server would be appreciated. I am basically interested in 

1. How to run a second module along with the first one, which would only care for streaming the data to the server?
2. How to pass the file location reference to the second module, is there a react-native way or I should do it in the native code?

P.S.: I am using the approach from here: https://medium.com/react-native-training/build-youtube-alike-livestreams-with-react-native-8dde24adf543
P.P.S.: I am planning to open a pull request once this is finished.

SergioGeeK7, 2018-12-16:
hey @AngelVenchev , I'm on the same path like you, I'm reading the file while is been recorded, the problem that I've noticed is there's a lag of 10 sec before the plugin dump the frames into the file, so I'm trying to get it to work from there, seems I'm stuck.

What I did to not block the UI was adding threading management...

```
- (dispatch_queue_t)methodQueue
{
  return dispatch_queue_create("com.facebook.React.my.cool.app", DISPATCH_QUEUE_SERIAL);
}
```
I'm on ios side for now cause it's the primarily platform the client care about

SergioGeeK7, 2018-12-16:
@lwansbrough Hi man, would you like to provide some code example or links about this approach you're outlining here?

`you'd need a way to control the stream so you can get the reference to it in another native module. You'd probably have to build the native module for streaming (unless one exists - but I doubt it) and that native module would read the native camera stream from this module`

thank you!

n1ru4l, 2018-12-18:
On `android` we could maybe pass something to the `RNCameraPackage` constructor? https://github.com/react-native-community/react-native-camera/blob/e79cd566ab61e31105dddef0193d7dd351a14d59/examples/basic/android/app/src/main/java/com/rncameraexample/MainApplication.java#L20-L35

On `iOS` we could handle the setup in `- (BOOL)application:(UIApplication *)application didFinishLaunchingWithOptions:(NSDictionary *)launchOptions`

https://github.com/react-native-community/react-native-camera/blob/e79cd566ab61e31105dddef0193d7dd351a14d59/examples/basic/ios/RNCameraExample/AppDelegate.m#L17-L25

AngelVenchev, 2018-12-18:
@n1ru4l Is this what you're suggesting an idea on general interoperability between separate react-native modules, or a solution to this issue. I am pretty new to react-native and I don't understand much, this is why I am asking.

AngelVenchev, 2018-12-18:
@SergioGeeK7 I am already thinking if running it in a separate component cannot work for some reason. I am thinking about implementing the streaming part inside the camera module, instead of throwing the base64 encoded pieces of the video to the JS thread directly flushing them to the server and exposing this to the props of the camera component, having this, supporting base64 and both OSs I will be willing to open a PR, but I want to try and stay away from native code making HTTP requests for now.

SergioGeeK7, 2018-12-19:
Hey @AngelVenchev that sounds good, wonder if flushing one event per frame with a base64 image will be a lot for the bridge and camera component, we won't know until you make it.

regarding my approach I ended up creating my own UI native component, connect to the camera stream and analyze each frame ( my app is about live object detection).

next, every time I detected the object, I flush an event to JS side.

I followed these links:

https://facebook.github.io/react-native/docs/native-components-ios
https://docs.opencv.org/2.4/doc/tutorials/ios/video_processing/video_processing.html

AngelVenchev, 2019-01-16:
I tried the described approach in [this tutorial](https://medium.com/react-native-training/build-youtube-alike-livestreams-with-react-native-8dde24adf543) which worked for android, however for iOS the streamed file was a bit different than the completed file. There were 6 bytes in the beginning of the file that specify how large is the video atom and how much it's offset from the beginning of the file which basically rends this format unstreamable. I guess with a lot more work on the native iOS side the format could be switched to something that's streamable. 
However our team doesn't have the capacity anymore. 

I've worked both on the [react-native-camera](https://github.com/angelvenchev/react-native-camera) and [rn-fetch-blob](https://github.com/angelvenchev/rn-fetch-blob) libraries. I've added an iOS implementation of a "Tailer" (tail -f) inside the react-native-camera lib if anyone is interested they could check out the forks of those repos in my account.

lake-effect, 2019-03-04:
How should we go about parsing frames from a stream? Guessing how many bytes to get and trying to track frames manually sounds like trouble...

Aubynj, 2019-03-17:
Any news on the streaming of the video directly from the camera recording? It seems this is getting pretty trouble for beginners in rn

WazzaJB, 2020-05-30:
Hey,

Sorry if this is the wrong place to post - I felt it was closely related to this issue.

I need to periodically retrieve the video stream (say every 30 seconds). It looks like if I were to do this, I wouldn't hit the same issues as a live stream due to not blocking the js bridge?

How would I go about this?

Thank you 😊

griffinshuth, 2020-06-11:
anyone have any ideas on this?  Im building a sports streaming app and need to be able to view the live camera and send that up to facebook live and youtube live

nachumFreedman, 2020-06-22:
for clarity can we describe what next the next steps to get this working, I could have a few developers working on this if we knew which direction to put them in.

n1ru4l, 2020-06-22:
@nachumFreedman Been a year since I worked with react-native and I don't know if there is an established best-practice regarding native module interop.

As I mentioned here:

https://github.com/react-native-community/react-native-camera/issues/135#issuecomment-448138511 

It might be possible to handle that kind of stuff by allowing to pass in an interface into the constructor/app setup for doing everything on the native side. Of course, someone has to come up with a smart API first. We could ship it as experimental first and then once we have more insight and use-cases we can then adjust it and make it stable.

I am happy to discuss/review pull requests if someone wants to implement this.

I think we can all agree that constantly (or even in intervals) passing video frames over the react-native bridge is something we should avoid nor advocate by providing a way to do so. If someone wants to go that route he might be better off forking react-native-camera.

closed 2020-08-28 by stale[bot]

coucoseth, 2023-05-18:
has anyone been successful with this