react-native-camera/react-native-camera #137: Delete taken photo

skeie, 2015-12-17:
Hi,

Based on [this](https://github.com/johanneslumpe/react-native-fs/issues/34), is it possible that delete photo will be implemented into react-native-camera?

lwansbrough, 2015-12-18:
Hey there, I have no plans to implement any fs operations in the camera module.

skeie, 2015-12-19:
Hi @lwansbrough, thanks for answering. Is it possible to save the images to a async storage instead of the disk? I would love to be able to not save each images to the disk.

lwansbrough, 2015-12-19:
AsyncStorage saves to disk in the background, so I'm not using it solves whatever problem you're trying to solve.

You also don't want to pass the image over the bridge in all likelihood, as it will lock the UI. Your best bet is to use the file system. If you really want to use JS APIs you can use the memory option as the target. This feature will come out of deprecation in 1.0.0.

sibelius, 2018-01-09:
CameraRoll can delete photos on react-native 52

closed 2018-01-09 by sibelius