react-native-camera/react-native-camera #138: stopCapture does not always invoke the capture method's callback immediately

rpastorelle, 2015-12-17:
Once calling `this.refs.cam.capture(options, bc)`, recording is requested to be stopped by invoking the `this.refs.cam.stopCapture();` method. If `stopCapture` method is invoked too soon after the `capture` call (<1s) then `stopCapture` never works.

Let me try to explain with code:

```
var options = {
  target: 'disk',
  mode: 'video',
  totalSeconds: 8,
}
this.refs.cam.capture(options, (err, path) => {
   // I am never called until totalSeconds is reached
});

// Call this very soon after capture begins... As if recording a 1/2 second video
setTimeout(() => {
  this.refs.cam.stopCapture();
}, 400);
```

lwansbrough, 2015-12-18:
Interesting, thanks for bringing this up. I'll add it to the list. I don't really have any leads on what may be causing this.

rpastorelle, 2015-12-18:
@lwansbrough I'm also unsure about that... I tried to dig into it a bit but wasn't able to get anything valuable. I know that it does make this call into the native API `NativeModules.CameraManager.stopCapture` below (but it has no affect):

```
  stopCapture() {
    if (this.state.isRecording) {
      NativeModules.CameraManager.stopCapture();
      this.setState({ isRecording: false });
    }
  }
```

I tried calling inside a `setInterval` instead of `setTimeout` to see if it would work eventually... but the callback was never triggered (unless the `totalSeconds` were supplied)

KanUchida, 2017-05-01:
+1 with camera

I created issue #688

Symous, 2017-05-11:
@rpastorelle Hi,my friend,I also meet this issuse , I try to stopCapture() 10 seconds after capture(),but the result is the video time is less than 10 second( 8 seconds actually).Do you have any idea about that now?

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco