react-native-camera/react-native-camera #141: Camera Component Freezes the app for a 1-2 seconds during Navigator transition..

gorkemyurt, 2015-12-31:
Interesting enough, this only happens when I disable chrome debugging (so maybe some sort of race condition)

I am going to spend some time and try to isolate the issue but here is a screencast of the problem; https://www.dropbox.com/s/tfhvv9u6f0epn0n/camera.mov?dl=0

gorkemyurt, 2015-12-31:
I guess navigator keeps the camera rendered all the time even if the component is not visible; handling this manually did the trick for now
```
componentWillUnmount() {
    this.setState({
      activeScene: false
    })
  }
```
and 
```
render() {
	  
    if (!this.state.activeScene) {
      return null;
    }
    return <Camera ... >
}
```

closed 2015-12-31 by gorkemyurt

alexleventer, 2016-06-28:
Thanks @gorkemyurt  - this saved me a lot of time!