react-native-camera/react-native-camera #143: User denied access, -3311

steveleec, 2016-01-06:
I have this code, like the example:

```javascript
export default class CameraLive extends React.Component {

  constructor(props) {
    super(props);
    this.state = {
      cameraType: Camera.constants.Type.back
    };
  }

  _switchCamera() {
    var state = this.state;
    state.cameraType = state.cameraType === Camera.constants.Type.back
      ? Camera.constants.Type.front : Camera.constants.Type.back;
    this.setState(state);
  }

  _takePicture() {
    this.refs.cam.capture(function(err, data) {
      console.log(err, data);
    });
  }

   _onBarCodeRead(e) {
    console.log(e);
  }

  render() {
    return (
      <View style={styles.container}>
        <Camera
          ref="cam"
          style={styles.cameraStyle}
          onBarCodeRead={this._onBarCodeRead}
          type={this.state.cameraType}>
          <TouchableHighlight style={styles.cameraButton} onPress={this._switchCamera.bind(this)}>
            <Text>The old switcheroo</Text>
          </TouchableHighlight>
          <TouchableHighlight style={styles.cameraButton} onPress={this._takePicture.bind(this)}>
            <Text>Take Picture</Text>
          </TouchableHighlight>
        </Camera>
      </View>
    );
  }
}
```
Once I clicked the button Take Picture, I got this message:

```javascript
Object {message: "Error Domain=ALAssetsLibraryErrorDomain Code=-3311 "User denied access" UserInfo={NSLocalizedFailureReason=The user has denied the application access to their media., NSLocalizedDescription=User denied access, NSUnderlyingError=0x7fe610905170 {Error Domain=ALAssetsLibraryErrorDomain Code=-3311 "(null)"}}"}
```

Any clue why this is happening?

rodrigoreis22, 2017-12-21:
I'm also receiving this message on iOS 11.2, iPhone 7. The user denied microphone permission but later enabled it on the app settings.. even that way we still get this error message.

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco