react-native-camera/react-native-camera #146: Rendering a camera in a tabbed view

ethul, 2016-01-07:
I may simply need to change my approach; however, I am wondering if there is a good way to render the same camera in two tabs. For example, I am looking to have one tab for still images and then a second tab for video. Consider the following example.

```javascript
/**
 * Sample React Native App
 * https://github.com/facebook/react-native
 */
'use strict';

const React = require('react-native');
const {
  AppRegistry,
  StyleSheet,
  Text,
  View,
} = React;

const ScrollableTabView = require('react-native-scrollable-tab-view');

const Camera = require('react-native-camera');

const ReactNativeCameraTabbed = React.createClass({
  getInitialState: function(){
    return {
      mode: Camera.constants.CaptureMode.still
    };
  },
  onChangeTab: function(tab){
    if (tab.i === 0) {
      this.setState({mode: Camera.constants.CaptureMode.still});
    }
    else {
      this.setState({mode: Camera.constants.CaptureMode.video});
    }
  },
  render: function() {
    const camera =
      <Camera ref="cam0"
              captureMode={this.state.mode}
              style={{ flex: 1
                     , backgroundColor: 'transparent'
                     , justifyContent: 'center'
                     , alignItems: 'center'
                     }}>
        <Text style={styles.text}>{this.state.mode}</Text>
      </Camera>

    return (
      <View style={styles.container}>
        <ScrollableTabView onChangeTab={this.onChangeTab}>
          <View tabLabel="Image"
                style={{flex:1}}>
            {camera}
          </View>
          <View tabLabel="Video"
                style={{flex:1}}>
            {camera}
          </View>
        </ScrollableTabView>
      </View>
    );
  }
});

const styles = StyleSheet.create({
  container: {
    marginTop: 20,
    flex: 1
  },
  text: {
    fontSize: 20,
    textAlign: 'center',
    margin: 10,
    color: 'white'
  }
});

AppRegistry.registerComponent('ReactNativeCameraTabbed', () => ReactNativeCameraTabbed);
```

The result is that the camera is only visible in the Video tab, and nothing shows up in the Image tab.

Any thoughts or points on whether or not this is possible would be most helpful.

lwansbrough, 2016-02-01:
Not sure how you could do this. There can only be one camera at a time. I think I would have to call this a design constraint of the device. You could try to detect when the tab is changing and swap out the component for a still image or an empty view, then switch it back. Not sure.

ethul, 2016-02-01:
Thanks for taking a look. I've tried a few ways to get this going, but no luck so far. I will have to dig into it further to see if there might a way to do this. I am open to ideas. Thanks again.

ethul, 2016-02-02:
Closing since it is not an issue, but a design constraint.

closed 2016-02-02 by ethul