react-native-camera/react-native-camera #149: Children components (visually) persist between umount and re-mount

monyxie, 2016-01-11:
When a `<Camera />` component is unmounted and mounted again, its children components will persist. 
[Demo](https://gist.github.com/monyxie/dd0552a72262fe508891):
[image]
Everytime `<Camera />` mounts, it renders a randomly positioned line as its child. You can see the previous lines persist when `<Camera />` umounts and mounts again.

MatthewRDodds, 2016-03-10:
I'm experiencing this same issue with iOS when trying to show a slightly different camera view (scanning barcode vs taking a picture):

[image]

The previous view is visually persistent, as if it's just been chilling in the background until I render my next camera view on top of it.

@monyxie were you able to find a workaround?

monyxie, 2016-03-11:
@MatthewRDodds We took our child component and put it under the same parent as `<Camera/>`, then we styled it so that it appears on top of `<Camera/>`.

MatthewRDodds, 2016-03-21:
That solution seems to have worked for me as a work-around.

rt2zz, 2016-03-21:
If I recall correctly, react-native-maps had a similar issue that they resolved. unfortunately I cannot find a reference in the commits. If anyway has any ideas help would be appreciated.

bz123, 2017-08-01:
I found a workaround, hacky but well I'll leave it here in case anyone else wants to hack around it.

The trick is to wrap your condition under a view within the camera itself e.g:

`<Camera><View> {this.state.isTrue && <Text> Hi </Text>} </View> </Camera>`

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco