react-native-camera/react-native-camera #150: How to use it scan barcode.

liukepeng, 2016-01-12:
run it I can saw camera open,but how to use it implement scan barcode? Or look like scanning something?

lwansbrough, 2016-01-12:
You can use the `onBarCodeRead` property on `Camera` to capture bar code reads. A bounding box is returned which can be used to draw the location of the barcode.

Talkwondo, 2016-08-26:
Hello,
Can you elaborate? maybe copy a text code here?

Thanks,

marxsk, 2016-09-21:
Look at an example on:
https://github.com/lazaronixon/react-native-qrcode-reader

sahir, 2017-09-24:
scan barcode example have a look : https://github.com/sahir/react-native-barcode-reader

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco