react-native-camera/react-native-camera #154: Crash with: "AVCaptureSession can't startRunning between calls to beginConfiguration / commitConfiguration"

benvium, 2016-01-16:
My client has reported a crash on iOS 9.2 (13C75) devices with the message "AVCaptureSession can't startRunning between calls to beginConfiguration / commitConfiguration".   
It's odd as I can't reproduce it on our own devices, including those running iOS 9.2

According to crashlytics, it strikes on `_32-[RCTCameraManager startSession]_block_invoke + 4295249440`

Looks like other people have had this issue when trying to start the capture in a background thread:
https://github.com/TheLevelUp/ZXingObjC/issues/55.

Any ideas?

benvium, 2016-01-16:
Ok, now I can reproduce this crash.

If I go into Settings / Privacy / Camera and deny app access, then the app crashes on the barcode scan page.

If I allow access again and restart the app, then it works.

benvium, 2016-01-26:
It looks like the intention is that the checkDeviceAuthorizationStatus method is used to check before running the camera. Unfortunately, this isn't available unless the <Camera> component is renderered, and if it renders - it crashes!

I solved this by pulling out the checkDeviceAuthorizationStatus method into a separate Native component and calling that before deciding to render the <Camera> component.

CameraPermission.h
```objc
#import <Foundation/Foundation.h>

#import "RCTBridgeModule.h"
@interface CameraPermission : NSObject <RCTBridgeModule>

-(void)checkDeviceAuthorizationStatus:(RCTResponseSenderBlock) callback;
@end
```

CameraPermission.m
```objc
#import "CameraPermission.h"
#import <AVFoundation/AVFoundation.h>

@implementation CameraPermission

RCT_EXPORT_MODULE();

RCT_EXPORT_METHOD(checkDeviceAuthorizationStatus:(RCTResponseSenderBlock) callback)
{
  NSString *mediaType = AVMediaTypeVideo;
  
  [AVCaptureDevice requestAccessForMediaType:mediaType completionHandler:^(BOOL granted) {
    callback(@[[NSNull null], @(granted)]);
  }];
}

@end
```

In my app..
```javascript
const CameraPermission = require('react-native').NativeModules.CameraPermission;

...
...

// don't render camera unless we have permission
        CameraPermission.checkDeviceAuthorizationStatus((err, isAuthorized)=>{
            if (isAuthorized) {
                this.setState({cameraAvailable: true});
            } else {
                AlertIOS.alert(
                    "Camera Access Denied",
                    "Go to Settings / Privacy / Camera and enable access for this app",
                    [{
                        text:"OK",
                        onPress:()=>{
                            this.props.navigator.pop();
                        }
                    }]
                );
            }
        });
```

lwansbrough, 2016-02-01:
So the authorization method has been pulled out into a static method now. It should be accessible via `Camera.checkDeviceAuthorizationStatus` - I think that will fix the issue you're describing.

benvium, 2016-02-01:
@lwansbrough great stuff, thank you. I will give it a try later

romansidko, 2017-04-14:
I have this bug too!

2017-04-14 14:18:29.070 XXX[1448:1266417] *** Terminating app due to uncaught exception 'NSGenericException', reason: '*** AVCaptureSession can't startRunning between calls to beginConfiguration / commitConfiguration'
*** First throw call stack:
(0x2440b91b 0x23ba6e17 0x29dc309f 0x2ec781 0x6f8b7f 0x703493 0x6fbde1 0x704ecf 0x7047cf 0x2413db29 0x2413d718)
libc++abi.dylib: terminating with uncaught exception of type NSException

Line [self.session startRunning]; in - (void)startSession method, RCTCameraManager file.

flybayer, 2017-04-15:
@romansidko check out #604

vvusts, 2017-10-27:
Is there any update on this issue?
I get the same error if I start stop camera view few times.
Is it possible that this is related to memory usage somehow?
I use react-native-permission to check if authorised and only then load camera view.
Issue is on ios 10...11.

manask88, 2017-11-14:
Hey @vvusts , https://github.com/lwansbrough/react-native-camera/pull/980 should fix the race condition problem which causes the error

closed 2018-01-12 by ospfranco