react-native-camera/react-native-camera #155: Cannot read property aspect of undefined.

colincgn, 2016-01-16:
I'm getting a the following error when trying to add this Camera example to my react app.

react-native-cli: 0.1.9
react-native: 0.18.0-rc

npm -v react-native-camera
3.5.3

However if I open up the node_modules and find the react-native-camera and open up the package.json I have this as the version ```"version": "0.3.8"```

colincgn, 2016-01-16:
Created a new project, followed steps again and it worked this time.  Safe to close this issue.

closed 2016-01-16 by colincgn

wulunyi, 2016-03-30:
After installing and linking package, important to rebuild native part of app.

jankalfus, 2016-07-14:
If you do the "Mostly automatic install with RNPM" for Android, you still have to do steps 2 and 4 of the manual install:

Open up `android/app/src/main/java/[...]/MainApplication.java (the Readme says MainActivity.java, but that's incorrect):
```
Add import com.lwansbrough.RCTCamera.*; to the imports at the top of the file
Add new RCTCameraPackage() to the list returned by the getPackages() method
```

Insert the following line inside the dependencies block in android/app/build.gradle:
```
compile project(':react-native-camera')
```