react-native-camera/react-native-camera #156: Next version?

JakeRawr, 2016-01-19:
Currently, the master branch has a fix for Tap to Focus crashing issue. But v0.3.8 doesn't have that fix.When will the next version be released? 

Thanks
-Jake

dangnelson, 2016-01-26:
Also, with the new Android support, will there be a new release?

lwansbrough, 2016-02-01:
Please pull the latest changes from master. There should be some much needed fixes in there.

lwansbrough, 2016-02-01:
I'll be pushing for `1.0.0` once APIs and functionality has stabilized.

viscasillas, 2016-02-01:
How does this Android support work? Can I use my existing code from iOS on Android?

lwansbrough, 2016-02-01:
Eventually the APIs will all work the same, for now you may have to make conditional assignments on the camera props because of lack of support of some features.

pocesar, 2017-03-02:
what the `next` branch is missing and/or update from 0.6?

sibelius, 2017-12-13:
we gonna work on the next release here https://github.com/lwansbrough/react-native-camera/issues/246#issuecomment-350577509

I think using cameraview from google would help maintaing android

closed 2017-12-13 by sibelius