react-native-camera/react-native-camera #158: Refocus when recording video

wgormley, 2016-01-21:
I am experiencing a visual refocus at the start of recording a video. here is my code if you want to try and reproduce it. the refocus usually makes the video very dark for a short time. any thoughts on what could be going wrong? i am running this on an 6S+ with 9.2.1

````
var Camera = require('react-native-camera');

var startVideo = false;
var recording = false;

class Full_Camera extends Component{
  constructor() {
    super();
    this.render = this.render.bind(this);
    this.state = { cameraType: Camera.constants.Type.back, captured: false };
  }
  render() {
    if (this.state.captured) {
      return <View></View>
    }
    else {
      return (
          <Camera
            ref="cam"
            style={styles.container}
            type={this.state.cameraType}
            defaultOnFocusComponent={false}
          >
            <TouchableWithoutFeedback onPressIn={this._buttonPressed.bind(this)} onPressOut={this._buttonUp.bind(this)}>
              <View style={styles.buttonsBorder}>
                <View style={styles.buttons}></View>
              </View>
            </TouchableWithoutFeedback>
          </Camera>
      );
    }
  }
  _buttonPressed() {
    startVideo = setTimeout(this._takeVideo.bind(this), 300);
  }
  _buttonUp() {
    clearTimeout(startVideo)
    if (recording){
      this.refs.cam.stopCapture();
      recording = false;
    }
    else{
      this.refs.cam.capture(function(err, data) {
        console.log(err, data);
      });
    }
  }
  _switchCamera() {
    var state = this.state;
    state.cameraType = state.cameraType === Camera.constants.Type.back
      ? Camera.constants.Type.front : Camera.constants.Type.back;
    this.setState(state);
  }
  _takeVideo() {
    recording = true;
    this.refs.cam.capture({mode: Camera.constants.CaptureMode.video}, function(err, data){
      console.log(err, data);
    })
  }
}
````

wgormley, 2016-01-28:
this is actually caused by the camera switching from the still sensor to the video sensor. i can add a delay in the code to keep the recording from starting until after that stutter happens. it happens even if i initially set the capture mode to video initially. is that initialization actually setting the sensor? ive only looked through the code a little, so any input would be great/

davidjeffries, 2016-02-20:
I noticed this as well... From what I can tell it is caused when adding the audio input to the capture session which happens here: https://github.com/lwansbrough/react-native-camera/blob/master/ios/RCTCameraManager.m#L596

I don't have much ios experience but I think adding the audio to the session causes the video to restart and refocus which looks bad in the resulting video file.

For now I hacked together a solution by modifying `initializeCaptureSessionInput` : https://github.com/lwansbrough/react-native-camera/blob/master/ios/RCTCameraManager.m#L373 to always add audio capture alongside the video capture, and I removed `[self initializeCaptureSessionInput:AVMediaTypeAudio];` from `captureVideo`. Not sure of any side effects to doing this but it seems to work.

gimlids, 2016-02-26:
Also seeing this... sometimes it also causes the video to start with the exposure way down: https://www.youtube.com/watch?v=RMID_ysDgJs

@davidjeffries one side effect might be always asking the user for permission to use the microphone. Perhaps an API should be added to allow specifying the capture mode at initialization time.

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco