react-native-camera/react-native-camera #160: Taking full resolution photos

tume, 2016-01-24:
When taking photos iPhone 6 the photo resolution seems to be 1080x1920 which is fine but not close to resolution it could (2448x3264). 

I was going through the source code to see if there is some kind of switch for quality but only thing I could find was this line:
https://github.com/lwansbrough/react-native-camera/blob/master/ios/RCTCameraManager.m#L126

Which uses AVCaptureSessionPresetHigh preset which if I understood correctly sets highest quality for video and not for stills. Switching this to AVCaptureSessionPresetPhoto allows me to capture full resolution photos.

I was wondering whether there was some specific reason to use PresetHigh instead of PresetPhoto? and is it possible to change it somehow from js side?

lwansbrough, 2016-02-01:
This was probably because we used to transfer data over the bridge and the full resolution would have been too slow. Would accept a PR to make this functionality possible (or perhaps even change the default)

viscasillas, 2016-02-01:
Stupid question problably but Can I use my existing iOS code on Android?? Do I need to use android studio? 

Thanks

pyrabbit, 2016-02-24:
I am interested in this type of enhancement and would consider working on a PR. I haven't done much with Objective-C or Swift or mobile for that matter. Is this a relatively straight forward fix or would it require a significant rehaul.

rpopovici, 2016-02-27:
It's done just needs to be merged https://github.com/lwansbrough/react-native-camera/pull/211

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco