react-native-camera/react-native-camera #162: Add timestamp to image captures in simulator

sharathprabhal, 2016-01-25:
Currently, image capture in the simulator is a plain white image. It will be helpful if the images had a timestamp to figure out if new captures were successful.

lwansbrough, 2016-02-01:
Do you think you could do something similar for Android?

sharathprabhal, 2016-02-01:
@lwansbrough Absolutely! I'll work on a PR soon.

lwansbrough, 2016-02-02:
Awesome!

adamkdean, 2016-08-22:
@sharathprabhal did you ever get round to this?

sharathprabhal, 2016-08-23:
@adamkdean Unfortunately, I did not get around to adding it for android.

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco