react-native-camera/react-native-camera #164: Linking does not seem to work

Mo-k-tec, 2016-01-26:
I followed all steps but when i build the android app i get the following error.

A problem occurred configuring project ':app'.
> Cannot evaluate module react-native-camera : Configuration with name 'default' not found.

Any ideas?

Regards,
Mo

dangnelson, 2016-01-26:
I'm having the same problem. 

I also don't see how to "Modify the ReactInstanceManager.builder() calls chain in android/app/main/java/.../MainActivity.java to include:" 

I don't have any references in my MainActivity.java file to ReactInstanceManager.

My path to MainActivity.java is also different:
    android/app/src/main/java/com/.../MainActivity.java

I running the latest version of RN.

dangnelson, 2016-01-26:
This issue is related to React-Native 0.18. Here's a related solution:
https://github.com/marcshilling/react-native-image-picker/issues/74

I'll try to get a working version from that.

dangnelson, 2016-01-26:
Tagging @farmisen since he was the original source for the documentation.

rpopovici, 2016-01-26:
I linked this project with "git+https://github.com/lwansbrough/react-native-camera.git" in package.json instead of using npm version ^0.3.8 and it works this way
It looks like the npm repository was not update to the latest version

ardowz, 2016-01-26:
In MainActivity.java

add this line:
import com.lwansbrough.RCTCamera.RCTCameraPackage;

before the class, that should do the trick.

dangnelson, 2016-01-26:
Worked. Thanks @ardowz 

Here's what my final MainActivity looks like:

    package com.myapp;
    
    import com.facebook.react.ReactActivity;
    import com.facebook.react.ReactPackage;
    import com.facebook.react.shell.MainReactPackage;
    
    import com.lwansbrough.RCTCamera.RCTCameraPackage; // IMPORT
    import java.util.Arrays;
    import java.util.List;
    
    public class MainActivity extends ReactActivity {
    
        /**
         * Returns the name of the main component registered from JavaScript.
         * This is used to schedule rendering of the component.
         */
        @Override
        protected String getMainComponentName() {
            return "myapp";
        }
    
        /**
         * Returns whether dev mode should be enabled.
         * This enables e.g. the dev menu.
         */
        @Override
        protected boolean getUseDeveloperSupport() {
            return BuildConfig.DEBUG;
        }
    
       /**
       * A list of packages used by the app. If the app uses additional views
       * or modules besides the default ones, add more packages here.
       */
        @Override
        protected List<ReactPackage> getPackages() {
            
          return Arrays.<ReactPackage>asList(
            new RCTCameraPackage(), // ADD
            new MainReactPackage()
          );
        }
    }

ardowz, 2016-01-26:
Nice. I managed to make mine to compile, but whenever I try to run it on my phone. I get an error which is in (https://github.com/lwansbrough/react-native-camera/issues/165) was wondering if you guys made it to work in android completely.

dangnelson, 2016-01-26:
Nope. Running into the same issue.

Mo-k-tec, 2016-01-27:
Also doesn't work here.

npm install react-native-camera@https://github.com/lwansbrough/react-native-camera.git --save
rnpm link react-native-camera

And then add the changes @ardowz said + @dangnelson

lwansbrough, 2016-02-01:
Hi all, please try with the latest commits from master.

awitherow, 2016-02-03:
any update on this? i'm going through the steps from @ardowz now. I've changed the config & I've reinstalled the latest via npm but I still get this output.

```bash
FAILURE: Build failed with an exception.

* What went wrong:
A problem occurred configuring project ':app'.
> Cannot evaluate module react-native-camera : Configuration with name 'default' not found.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

BUILD FAILED
```

mkraft, 2016-02-03:
@rpopovici Worked for me, thanks.

derdav3, 2016-09-13:
I solved it like this..
instead of importing:  
`import com.lwansbrough.RCTCamera.*;`
do this:  
 `import com.lwansbrough.RCTCamera.RCTCameraPackage;`

closed 2016-09-13 by rt2zz