react-native-camera/react-native-camera #165: Getting RCTCameraViewManager cannot be cast to com.facebook.react.uimanager.ViewGroupmanager

ardowz, 2016-01-26:
Using the example converted to ES6 the code works for iOS but when tried to be used on android, that error message is received.

I'm also using the most current git commit to implement it on android

dangnelson, 2016-01-26:
Yeah I'm getting this too.

dangnelson, 2016-01-26:
This seems like a related issue:
http://stackoverflow.com/questions/32704336/how-to-access-activity-from-a-react-native-android-module

I think the component needs to be modified to pass in the Activity.

ardowz, 2016-01-26:
I'm trying to implement whats in that document, but i dont know what to do with the activity once i get it in to RCTCameraViewManager

dangnelson, 2016-01-28:
I was able to get the error to go away by changing RCTCameraViewManager from 

> public class RCTCameraViewManager extends SimpleViewManager<RCTCameraView>

to 

> public class RCTCameraViewManager extends ViewGroupManager<RCTCameraView>

as mentioned here:
http://stackoverflow.com/questions/35040265/react-0-18-causes-module-error/35061798#35061798

Working on device testing now to see if it works...

dangnelson, 2016-01-28:
That seemed to fix the error, but no luck on actually getting it to work.

ardowz, 2016-01-29:
Using what you mentioned and added some entries in AndroidManifest.xml

`<uses-permission android:name="android.permission.CAMERA" />
    <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE"/>
    <uses-feature android:name="android.hardware.camera"
                  android:required="true"/>
    <uses-feature android:name="android.hardware.camera.autofocus" />`

after 
`<uses-permission android:name="android.permission.INTERNET" />`

I got the camera to work. I was checking out a different library and tried to point the dots.

dangnelson, 2016-01-29:
Just to review.. I still can't get this to work in RN 0.18.

Here are the changes I've made so far:

Made the changes in the docs to these files:
android/settings.gradle:
android/app/build.gradle

Then attempted these to fix the errors:

**MainActivity.java**

```
import com.lwansbrough.RCTCamera.RCTCameraPackage;  // <--- import

public class MainActivity extends ReactActivity {
  ......

  /**
   * A list of packages used by the app. If the app uses additional views
   * or modules besides the default ones, add more packages here.
   */
    @Override
    protected List<ReactPackage> getPackages() {
      return Arrays.<ReactPackage>asList(
        new RCTCameraPackage(), // <------ add here
        new MainReactPackage());
    }
}
```

**AndroidManifest.xml**

```
<uses-permission android:name="android.permission.INTERNET" />
// add these below above line
<uses-permission android:name="android.permission.CAMERA" />
<uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE"/>
<uses-feature android:name="android.hardware.camera" android:required="true"/>
<uses-feature android:name="android.hardware.camera.autofocus" />
```

**Changed RCTCameraViewManager**
`public class RCTCameraViewManager extends SimpleViewManager
`
to
`public class RCTCameraViewManager extends ViewGroupManager
`

ardowz, 2016-01-29:
Not sure if this will work, but based on the other library (react-native-image-picker), but i added this in my MainActivity.java
`import android.content.Intent;`

before the Facebook imports.

Its been working on my phone already

BlackGlory, 2016-01-30:
As @ardowz said, my phone and emulator work fine after add `import android.content.Intent`.
Maybe this project's Docs and Code need to be updated for react-native 0.18.

sibelius, 2017-12-09:
this is fixed on master

`ViewGroupManager` is needed to handle ui component with children

closed 2017-12-09 by sibelius