react-native-camera/react-native-camera #167: Camera on android has no preview

rpopovici, 2016-01-27:
I managed to build with camera for android, but no preview is available. Do you guys have any idea why?

rpopovici, 2016-01-29:
Guys, any idea how to see the camera preview on android?

smallsaucepan, 2016-01-29:
What do you see @rpopovici? Can you provide some other details about your app and environment? I may be able to help having been in a similar position to you earlier today.

rpopovici, 2016-01-29:
Yes, I am building with react native 0.19.0-rc on Genymotion. I did all the suggested fixes like switching to ViewGroupManager and added user permisions. Anyway, the camera module loads, but I see no preview(transparent background). If I take a picture it works on the emulator, but on the device I get this error: 
[image]

rpopovici, 2016-01-30:
I discovered why the camera on android it's not showing preview. Apparently RCTCameraViewManager.setType() is never called. If I add setCameraType(2); in RCTCameraView constructor I can see the preview. Now, the question is: why settings JS props don't call the native counterparts in java code?

rpopovici, 2016-01-31:
This is related to the fact that @ReactProp annotated props are never called. Any idea why?

dangnelson, 2016-02-01:
Using a brand new RN 0.19 project and installing from:

> npm install react-native-camera@https://github.com/lwansbrough/react-native-camera.git --save

and following the new instructions in the docs I get no camera preview and the following error when hitting capture:

 > couldn't find key type in dynamic object

@rpopovici Can you give an exact example of where you added setCameraType(2); ?

rpopovici, 2016-02-01:
The latest incarnation of react-native-camera is totally busted with rn 0.19 on android. 
You have to do the following to make it work:
1 Include View and add ...View.propTypes under propTypes in Camera.js 
2 Rename REACT_CLASS in your android RCTCameraViewManager to  REACT_CLASS = "RCTCamera";  in order to match the iOS name.
3. Add setCameraType(2); setAspect(0); at the end of your RCTCameraView constructor. This is needed because ReactProps annotations are ignored for some reason..

lwansbrough, 2016-02-01:
Can you guys submit a pull request if you've managed to fix some functionality? Thanks!

rpopovici, 2016-02-02:
@lwansbrough done!

dangnelson, 2016-02-02:
This is the first fix where I've seen the Camera working, but the Capture button isn't visible over top. I'm using a new project and the default example.

lwansbrough, 2016-02-02:
Interesting. That should be a relatively easy fix - just need to make sure sub views are being added correctly. Just going to take a look at the PR now.

lwansbrough, 2016-02-02:
Going to take a closer look at this later.

rpopovici, 2016-02-02:
Yes, adding children to Camera does not work for now. You can add your image with position absolute outside camera tag.

agrass, 2016-02-10:
I updated to react native 0.19 and I'm using @rpopovici fixes. I see the preview perfect, but when I press the capture button I get an error. Anybody knows why could happen this?

```
 Exception in native call from JS
unknown:React: com.facebook.react.bridge.NoSuchKeyException: couldn't find key type in dynamic object
unknown:React: 	at com.facebook.react.bridge.ReadableNativeMap.getInt(Native Method)
unknown:React: 	at com.lwansbrough.RCTCamera.RCTCameraModule.capture(RCTCameraModule.java:154)
unknown:React: 	at java.lang.reflect.Method.invoke(Native Method)
```

closed 2017-04-26 by ptomasroos