react-native-camera/react-native-camera #169: How tap-to-focus feature work

JeremDsgn, 2016-01-27:
Hi!

I would like to let the user focus to take picture. But I don't know how it works.

I found on the documentation the section related to the problem : https://github.com/lwansbrough/react-native-camera#defaultonfocuscomponent But is there any example to view how it works ? I don't have any idea on how to implement this. With `defaultOnFocusComponent ` at `true` I didn't have **default internal implementation of visual feedback for tap-to-focus gesture**

Someone can help me on this point ?

realjamie, 2016-01-27:
Hi,

Try set defaultOnFocusComponent to true and onFocusChanged={this._handleFocusChanged}. In your callback, you can do:

  _handleFocusChanged(event) {
      console.log(event.nativeEvent.touchPoint);
      // call capture() here
  }

The visual feedback is already provided in this component. 
Hope this helps!

JeremDsgn, 2016-01-27:
Hum, it breaks the app. The focus seems to be triggered, but an error occured.

`*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '*** -[__NSPlaceholderDictionary initWithObjects:forKeys:count:]: attempt to insert nil object from objects[0]'
libc++abi.dylib: terminating with uncaught exception of type NSException
(lldb) `

realjamie, 2016-01-28:
Maybe the error occurred because of capture()? Try remove capture() and only print log first?

wgormley, 2016-01-28:
this fails for me too, i dont have capture in my code, just attempting to console log the event and it fails with the error @JeremDsgn gets.

nitelite, 2016-01-29:
I am trying to get focus on tap to work on Android, and I am also lacking the focus widget and the event does not fire. I have tried both with defaultOnFocusComponent set to true and false, and both with and without the onFocusChanged event bound. No focusing in any of these configurations. Anyone got a suggestion of where to attack this problem? 

I am running the latest code from the git repo on actual device (Nexus 5X) with Android 5.0.1.

JeremDsgn, 2016-01-31:
Some help @lwansbrough ?!

nitelite, 2016-01-31:
My goal was to use it for scanning barcodes, and after looking through the code a bit it looks like this project doesn't support barcode scanning on Android. I switched to react-native-barcodescanner for the Android part of the application and that seems to work fine, so no hurry on my behalf :)

FrankSalad, 2016-01-31:
What version of react-native are you on @JeremDsgn - I'm getting the same error on react-native 0.14.2 react-native-camera 0.3.8

JeremDsgn, 2016-01-31:
I'm on react-native 0.19.0 and react-native-camera 0.3.8.

FrankSalad, 2016-01-31:
@JeremDsgn did some digging, looks like it was fixed in November. https://github.com/lwansbrough/react-native-camera/issues/107

 I don't know why our npm must have pulled in an old revision.

FrankSalad, 2016-01-31:
@lwansbrough might be useful to bump the version (not sure how to force npm to see the change)

FrankSalad, 2016-02-01:
@JeremDsgn I had to `npm uninstall react-native-camera` and then `npm install https://github.com/lwansbrough/react-native-camera.git --save` to get the revision with the fix.

lwansbrough, 2016-02-01:
IMO this API needs to change. I think a `renderTapToFocus` prop would be appropriate or something similar - I want to allow for customization of that component somehow. If anyone wants to try doing this now would be a good time as I'm working through the bugs and upgrading APIs.

JeremDsgn, 2016-02-01:
@FrankSalad Ok, that's it ! Cool ! The last version of master is fixing the problem.
@lwansbrough Could you bump the version ? To avoid to download the dependencie from repository master.

Is there any way to custom the style of the focus square ? Changing the `borderWidth`, `borderColor` ?

lwansbrough, 2016-02-01:
@JeremDsgn Won't be version bumping until the API is ready. There are major changes that are unfit for a minor version bump.

lwansbrough, 2016-02-01:
See my previous comment with regards to what needs to be done about the focus square.

JeremDsgn, 2016-02-01:
Oh sorry, read to fast on the early morning, and you answered all of my question. The basic feature works. Hope to see the customization soon !

closed 2016-02-01 by JeremDsgn