react-native-camera/react-native-camera #173: android.hardware.Camera is deprecated

Phoenix1708, 2016-01-31:
Just a heads up. The native camera library used ("android.hardware.Camera" is deprecated)[http://developer.android.com/reference/android/hardware/Camera.html]. Device running on API level 21+ will experience subtle behaviour if they trying to use this library. Therefore, more backwards compatibility work needed for the Android support I guess. I am looking into it at the moment.

corbt, 2016-02-05:
It's marked as deprecated because it has been replaced by the camera2 library on API 21+. However, it is still fully supported and you won't run into any issues using it. The vast majority of camera apps in Android are still using it. Since camera2 isn't available for API < 21 and React Native supports API 16+, using the "deprecated" Camera module is probably the right choice for this project, as it is for any project wanting to support older versions of Android.

sibelius, 2017-12-10:
check this https://github.com/lwansbrough/react-native-camera/issues/246

we should go with CameraView

sibelius, 2018-01-05:
let's track this here https://github.com/react-native-community/react-native-camera/issues/246

closed 2018-01-05 by sibelius