react-native-camera/react-native-camera #174: When I switch to the settings, and then switch camare permissions, my app always crash

linshaolie, 2016-02-02:
[image]

mikollo, 2016-02-08:
https://github.com/lwansbrough/react-native-camera/issues/134

You cant render <Camera /> if You detect that user disabled acces for camera 😉

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco