react-native-camera/react-native-camera #175: Meet some issues running in android

soliury, 2016-02-02:
I have meet some issues running in android, I use Samsung Galaxy s6 API 23 in Genymotion:

```
Warning: Native component for "RCTCamera" does not exist
```
And it can run, but , the camera view is empty, I mean it have nothing, just white.

lwansbrough, 2016-02-02:
Thanks for reporting. This is an issue that's being tracked in multiple places. I'll try and have a fix tonight (8+ hours away from now.)

Phoenix1708, 2016-02-03:
You are very likely used npm to installed the out of date version i.e not what's on the master branch @soliury see #177 it will probably give you an idea.

@lwansbrough I guess the the README needs to be updated now otherwise there will be more such issue raised.

soliury, 2016-02-04:
@lwansbrough @Phoenix1708 I have install the latest master code, but failed.  I mean, when I meet the issue, I have installed the latest code from master, and then show:

```
Warning: Native component for "RCTCamera" does not exist
```

And today, I reinstall the package from the new master, but failed. 

[image]

Phoenix1708, 2016-02-04:
Hi @soliury , do you mind to show me how did you installed the library ?

soliury, 2016-02-04:
Just like this:

```
npm install react-native-camera@https://github.com/lwansbrough/react-native-camera.git --save
rnpm link react-native-camera
```

And then, I check the code, it have linked.

Phoenix1708, 2016-02-04:
If you put ```"react-native-camera": "git+https://github.com/lwansbrough/react-native-camera.git"``` in your package.json as one of the dependencies and run ``` npm install ``` you should be able to get the correct version and the required file.

lwansbrough, 2016-02-04:
I've had problems with rnpm setting the include path of the native modules in settings.gradle. Make sure these paths are set correctly.

kwngo, 2016-02-11:
Does react-native-camera only work with devices running API 23?

benjamin-wss, 2017-06-14:
Heya, had a similar issue today when using it. Apparently, the react-native link command did not update the gradle file here : `./android/app/build.gradle`

I had to add the `compile project(':react-native-camera')` myself inside, then it worked.

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco