react-native-camera/react-native-camera #177: Using react-native-camera for android on Ubuntu

cubbuk, 2016-02-02:
I could't run my app using the camera component on ubuntu. 

Ubuntu version: 14.04
React-native: 0.18.1
react-native-camera: 0.3.8

I installed the latest version using the given command:

> npm install react-native-camera@latest --save

Strangely I saw that the java package name in the installed library is com.lwansbrough.ReactCamera whereas the lastest version on master branch is RCTCamera. So seems like I am downloading an older package.

The manifest file I got after npm install is the following

```
<manifest xmlns:android="http://schemas.android.com/apk/res/android" 
package="com.lwansbrough.ReactCamera"> //ReactCamera instead of RCTCamera seems to be suspicious
  <uses-permission android:name="android.permission.CAMERA" />
  <uses-feature android:name="android.hardware.camera" />
  <uses-feature android:name="android.hardware.camera.autofocus" />
</manifest>
```

So when I tried to run my application I am getting the following error: 

A problem occurred configuring project ':app'.
> Cannot evaluate module react-native-camera : Configuration with name 'default' not found.

Have any idea how to solve it?
Thanks.

gbuszmicz, 2016-02-02:
+1

> A problem occurred configuring project ':app'.
> Cannot evaluate module react-native-camera : Configuration with name 'default' not found

Same error here but I´m using Mac

tarikmlaco, 2016-02-02:
I managed to get past the 
'Cannot evaluate module react-native-camera : Configuration with name 'default' not found.'
You just need to pick up the *latest* version, not from npm, but from git.
As the author noted on another (similar) issue, you need to type
npm install react-native-camera@[git link goes here] --save
so that you *actually* get the latest version.

After that, you should get into another problem...
Warning: Native component for "RCTCamera" does not exist
(React Native Debugger - Google Chrome)

Anyone with an idea what to do next?

lwansbrough, 2016-02-02:
@tarikmlaco I'm gonna try and get the Android version working at least partially working today.

tarikmlaco, 2016-02-02:
Ok. I have been forced to use 'react-native-camera-android', and it still
doesn't work as expected, but at least it builds properly. I am sure it is
a fork of your project, but I think it's worth checking out, and maybe even
merging later in the future.
Keep up the great work. Cheers :)

zackify, 2016-02-03:
Getting the same thing here. `Cannot evaluate module react-native-camera : Configuration with name 'default' not found.` I really need this to work, wish I knew Java enough that I could help

tarikmlaco, 2016-02-03:
@zackify you just need to pull the latest branch, NOT from npm, but from git.
npm install react-native-camera@[git link goes here] --save

But you won't be able to build the module, until there's a working android version available.
In the meantime, there's a (partially) working fork, react-native-camera-android, i suggest you also pick the latest version from GitHub, if you're really in a hurry.

Hope this helps.

Phoenix1708, 2016-02-03:
@zackify @tarikmlaco  @gbuszmicz I got the Android version built and running. I think if you look into the "android" folder of your installed version of this library i.e in node_module, you probably won't find any ```build.gradle``` file, which means you are building a project which doesn't know how to build itself. Other than wait for the working version to be release to npm you can folk this repo and [reference it in your package.json](http://www.devthought.com/2012/02/17/npm-tricks/). You will probably hit [some other issues](https://github.com/lwansbrough/react-native-camera/issues/165) which has already been resolved by someone else. Look carefully [at the end of this issue](https://github.com/lwansbrough/react-native-camera/issues/165) you will find solution.

Nevertheless, we can wait for @lwansbrough to release the working version and thanks @lwansbrough for creating this library.

And one last thing, in case your problem is resolved: You're welcome :)

Phoenix1708, 2016-02-03:
@lwansbrough as I mentioned in #173 we will have to resolve this or more issues will be raised regarding to various strange behaviour of the Android camera.

nithin-c, 2017-06-17:
In package.json under dependencies give 
`"react-native-camera": "git+https://github.com/lwansbrough/react-native-camera.git",`
run `npm install`

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco