react-native-camera/react-native-camera #178: Tap To Focus question

rt2zz, 2016-02-02:
Does `defaultOnFocusComponent` just disable visual feedback, but preserve the tap to focus functionality?

I am looking into potential android implementation, but am not sure what controls to expose via props, and what the default behavior should be.

lwansbrough, 2016-02-02:
I really dislike this API - I'm going to remove it for `1.0.0` - I would like to have a property `renderFocusArea` which accepts a function that accepts coordinates points, and leaves the user to return an instantiated component which will be shown at those coordinates. Like so:

```javascript
<Camera renderFocusArea={(coords) => <MyFocusBox style={{top: coords.x - 50, left: coords.y - 50, height: 100, width: 100}}/>}/>
```

Coords would be relative to the size of the camera preview, so there wouldn't be any translation required for the user.

lwansbrough, 2016-02-02:
I think then that there wouldn't need to be a native API for the focus area, just an event that is triggered when the focus area is changed, with coordinates attached. Then we can handle that event inside the Camera component.

rt2zz, 2016-02-02:
I like that. I would propose we should also:
- default to a reasonable tap to focus region
- allow disabling tap to focus either by setting renderFocusArea={false} or exposing a separate prop

lwansbrough, 2016-02-02:
I think tap to focus enabled by default makes sense, so in that case `renderFocusArea={false}` would make sense, yeah.

The same could be done for face detection where a bounding box and coordinates are sent back to a `renderFaceDetectionArea(coords, size)` where coords are now the top left edge of the region, and size is the size of the region.

lschmierer, 2016-08-18:
Tap To Focus is not implemented in Android at all, right? And on iOS?

lazarte, 2016-09-01:
Below code work on IOS but not on android?

```
<Camera
    onFocusChanged={this.onFocusChanged.bind(this)}
    defaultOnFocusComponent={true}
    ref = {(cam) => {this.camera = cam}}
    style = {styles.preview}
    captureAudio = {false}
    type = {Camera.constants.Type.back}>
</Camera>
```

ivosabev, 2017-03-15:
Version 0.6.0 the default tap-to-focus doesn't work in Android.

LEEY19, 2017-04-20:
@lazarte hi, do you know if the code you posted already works on android?

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco

ahsan-storm, 2018-10-16:
Please reopen this issue

ahsan-storm, 2018-10-16:
Tap to focus still not working in android