react-native-camera/react-native-camera #181: Uncaught Error: undefined is not an object (evaluating 'CameraManager.Aspect')

tomprogers, 2016-02-03:
I'm getting this error on simulators and on my real iPhone 5.

I followed the "mostly automatic" instructions. None of the commands threw any errors.

I'm building an app using ExponentJS, in case that's relevant. I haven't figured out how to follow the manual steps yet (I don't have an Xcode project).

tomprogers, 2016-02-08:
The problem appears to be related to the fact that projects created with ExponentJS don't come with all the Xcode and Objective-C stuff that you get when you create a new project via `react-native init [myappname]`. In this scenario, the rnpm link command listed in the README performs no work -- and shows no errors -- and this module won't function.

I don't think this project is responsible for resolving the issue, but it does seem like this module would benefit from some more aggressive error logging. I got lucky by thinking to build a new react-native project and integrate this component; at no point did this module indicate that it's dependencies weren't satisfied.

closed 2016-02-08 by tomprogers

shssoichiro, 2016-03-24:
For those who may still be googling to find the answer to this, my issue was that I had not restarted XCode/rebuilt the project after installing react-native-camera. Reloading the JS isn't enough, I had to exit the simulator and restart XCode so that the native modules would build.

Hopefully this helps someone.

jorry, 2016-05-15:
android +1

KayahanKahriman, 2016-07-30:
Im still getting this error. rnpm didnt work, tried to install manually but it didnt work either. Closed react-native packager and closed emulator (geny motion - galaxy s6) none of these didnt work. Plus, this should have been explained in more detail i think because there's no "getPackages method" there.

"Add new RCTCameraPackage() to the list returned by the getPackages() method"

simzen85, 2016-08-29:
I'm having this error with ExponentJS, does anyone know how to solve this ? Thanks all

Anuj-786, 2016-09-07:
hey, I got the same error in android simulator. whats is this error and how to sole it please anyone help.

hbd, 2016-12-04:
Getting this error on 
```
react-native-cli: 1.2.0
react-native: 0.28.0
```

stoefln, 2017-03-20:
strange thing is I am getting this error in a project where I am not even using this library :/
In my camera project everything works fine.

stoefln, 2017-03-20:
Restarting the react server helped. Error gone!