react-native-camera/react-native-camera #182: Reject function error 'Too many arguments to block call, expected 1, have 3

evanmrose, 2016-02-05:
`else if (target == RCTCameraCaptureTargetCameraRoll) {
    [[[ALAssetsLibrary alloc] init] writeImageDataToSavedPhotosAlbum:imageData metadata:metadata completionBlock:^(NSURL* url, NSError* error) {
      if (error == nil) {
        resolve([url absoluteString]);
      }
      else {
        reject(RCTErrorUnspecified, nil, RCTErrorWithMessage(error.description));
      }
    }];
    return;
  }
`

The reject function returns 'Too many arguments to block call, expected 1, have 3. The same thing happens for self.videoReject calls. 8 errors in total.

The errors occur in RCTCameraManager.m

evanmrose, 2016-02-05:
Looks like RCTPromiseRejectBlock now only accepts 1 argument. I'm in RN 0.19. I'm not sure what version your supporting up to now but let me know if I should submit a PR fixing this by passing only the RCTErrorWithMessage to the reject fn. @lwansbrough

corbt, 2016-02-05:
Actually, RCTPromiseRejectBlock recently moved from 1 to 3 arguments (https://github.com/facebook/react-native/commit/0c5f279c9df869db2afb3e1d92a54e6d930551ad), so this code is in line with the new syntax. But that change was prior to the release of 0.19, so I'm not sure why this wouldn't work with that version. :/

evanmrose, 2016-02-05:
Not sure what's up or if my version is off but what I posted fixed my issue...

\_(ツ)_/¯

closed 2016-02-05 by evanmrose