react-native-camera/react-native-camera #185: White screen

juggleru, 2016-02-07:
I'm using sample from readme and see just red bordered white screen with small area with text [CAPTURE].

React-native 0.19

ananogal, 2016-02-07:
I get exactly the same.

agrass, 2016-02-10:
I'm with the same problem

agrass, 2016-02-10:
Hey guys, check this issue #167. It's there a pull request of @rpopovici that solve it

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco