react-native-camera/react-native-camera #188: [Enhancement] Returning video/picture metadata

corbt, 2016-02-08:
It's often useful to get metadata about a picture/video you've captured. It's especially useful to know a video's duration. This can be estimated by manually timing the user's button presses, but it would be easier and more accurate if the framework did this automatically.

I'd like to propose a change to the `capture` method's resolved return value. Currently, it just returns the file's path. This could be improved by returning a `status` object, with the following properties:

Photos: `path`, `width`, `height`, `file_size`
Videos: All of the above, plus `duration`.

This change would also make the library more future-proof, because it would be easy to add more properties if necessary.

Happy to put together a PR if this looks good. Thoughts?

aforty, 2016-03-03:
I would find this useful. 👍

corbt, 2016-03-04:
@lwansbrough have you had a chance to think about this?

jadnco, 2016-03-13:
Adding a `location` object that includes latitude and longitude would also be awesome.

sscaff1, 2016-03-30:
This is exactly what I need. I need the metadata on the callback as @corbt mentioned above. Has any work been done on this?

corbt, 2016-03-30:
No, I don't think anyone is working on this. I'm happy to implement this in a PR though if @lwansbrough agrees it's a good idea. It's a breaking change so consensus/approval is important.

rt2zz, 2016-03-30:
This would be great to have, and since we are looking at a putting together 1.0 release, breaking api changes are very much fair game. 

Also with the promise based api, we do not have a lot of alternative options to support metadata.

:+1:

yodacom, 2016-03-30:
👍

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco