react-native-camera/react-native-camera #189: $300 Bounty for cross-platform video support

Dyscrete, 2016-02-09:
I'm offering 0.5 bitcoin (approx $300 at the time of this writing) for working Android & iOS video support

hunteredwards, 2016-02-12:
@Dyscrete I have a working implementation of video recording that I built on top of a fork of this repo by @timmh. I'm going to get this repo working for RN 0.19, then fork it, and hopefully, I can have something up by the end of the weekend.

Dyscrete, 2016-02-12:
@hunteredwards I'm using RN 0.19 so it would be awesome if you could do this and I'd happily pay the bounty!

JBerendes, 2016-02-24:
To be clear, it is known that iOS is currently working. I'm using it and that works fine. This is a request to get the feature working on Android. Feature parity across platforms is ideal. Thanks to anyone who takes this on.

klvs, 2016-03-07:
Any update on this?

elaygl, 2016-03-22:
+1 video for android

Dyscrete, 2016-03-25:
Can we get this going?!

vespakoen, 2016-04-15:
My initial stab at it here:

https://github.com/vespakoen/react-native-camera/commit/3d0ee9475166d83d9bdb73dfb4cf3ba681e3dd48

I feel I have tried anything, but cannot get a playable video file (there is data in there for sure, it just doesn't play, not on android, not in VLC)

This might be due to my device(s), the Samsung Galaxy S4 Mini, and a Nexus 7.

I commented a lot of the code to hopefully get other people going on trying it out / play with it / report back with findings / results.

To try out my branch, run:

```shell
npm install --save git+https://github.com/vespakoen/react-native-camera.git#master
```

Helpful links:

http://developer.android.com/guide/topics/media/camera.html#capture-video
http://developer.android.com/reference/android/media/MediaRecorder.html
http://developer.android.com/reference/android/media/CamcorderProfile.html

marcejohnson, 2016-04-16:
I have been working on this for awhile, and my fork is ready for a pull request:
https://github.com/lwansbrough/react-native-camera/pull/262

- Supports same parameters and values as IOS, including captureMode, captureTarget, captureQuality, etc.
- Updated Readme.md
- Example code

Goal is "drop in" for compatibility with IOS... to make Android and IOS equivalent. I will continue to enhance this as bugs or additional changes are needed since I need this for my own project.

Dyscrete, 2016-04-18:
Thank you @marcejohnson and @vespakoen, I'll take a look at it shortly

marcejohnson, 2016-04-27:
Got in all the improvements and features requested so far.  Runs well in my test apps. Still no conflicts on base branch.  Grab the latest version and let me know what you think.

marcejohnson, 2016-08-29:
@samhunta the pull request is officially merged and now part of the repo:
https://github.com/lwansbrough/react-native-camera/pull/262

You can close this.

closed 2017-05-23 by samhunta