react-native-camera/react-native-camera #190: How to define Barcode scanner scan area?

rohitgoyal, 2016-02-09:
Hi
Currently there is not mention of the scan area and by default it is in the middle so I have to do guessing to scan a barcode. Is it possible to make the whole screen scannable? 

Also is scanning feature not available for Android yet?

Thanks,
Rohit

kwngo, 2016-02-11:
Also need this functionality :+1:

alexdunae, 2016-04-11:
In my experiments on iOS, QR codes will scan anywhere but 1D barcodes need to be within the `rectOfInterest`.

[Here are the docs from Apple](https://developer.apple.com/library/ios/technotes/tn2325/_index.html#//apple_ref/doc/uid/DTS40013824-CH1-WHY_IS_MY_1_DIMENSIONAL_BARCODE_NOT_DETECTED_AT_ALL_LOCATIONS_IN_THE_IMAGE_).

E-P1983, 2016-05-16:
+1, would love to see the scanning area for 1D barcodes exposed so we can adjust it. Would be a tough one though.

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco