react-native-camera/react-native-camera #193: Error capture method in Android: NoSuchKeyException: couldn't find key type in dynamic object

agrass, 2016-02-11:
When I try to use the capture() method I get this error (Same with type, title and description): 

``` 
Exception in native call from JS
unknown:React: com.facebook.react.bridge.NoSuchKeyException: couldn't find key type in dynamic object
unknown:React:  at com.facebook.react.bridge.ReadableNativeMap.getInt(Native Method)
unknown:React:  at com.lwansbrough.RCTCamera.RCTCameraModule.capture(RCTCameraModule.java:154)
unknown:React:  at java.lang.reflect.Method.invoke(Native Method)
```

kwngo, 2016-02-11:
Getting a similar error running react-native 0.19

agrass, 2016-02-11:
Yoy can solve it adding ```  type: props.type, ``` into capture method default options in Camera.js or adding in the method capture like this:

```camera.capture({type: 2, title: 'example_title', description: 'example_description'});
```

kwngo, 2016-02-11:
Thanks, think it worked. I'm getting content://media/external/images/media/22 events in my logs.

agrass, 2016-02-15:
fixed in PR #192 with default params

closed 2016-02-15 by agrass