react-native-camera/react-native-camera #194: [Android] Switching to front camera doesn't work

hunteredwards, 2016-02-12:
Has anyone been able to implement a working switching between the back and front cameras for Android?

Only the back camera initializes the camera view, and with it, capture works. 

However, the front camera renders nothing, and the capture method stops working.

Device: Samsung Galaxy S5
RN: 0.19
Android: 5.1.1

cubbuk, 2016-02-22:
Same issue using Nexus 5. When trying to switch to front came screen freezes with the current snapshot of the screen, but app stays responsive.

RN: 0.20
Android: 6.0.1

cubbuk, 2016-02-22:
Seems like this is fixed in the latest version of repo thanks to @rpopovici. 
The below check was required for the front camera to work properly.
```
List<String> focusModes = parameters.getSupportedFocusModes();
if (focusModes.contains(Camera.Parameters.FOCUS_MODE_CONTINUOUS_PICTURE)) {
   parameters.setFocusMode(Camera.Parameters.FOCUS_MODE_CONTINUOUS_PICTURE);
}
```

hunteredwards, 2016-02-23:
@cubbuk thanks for finding that. You're a life saver!

closed 2016-02-23 by hunteredwards