react-native-camera/react-native-camera #197: takePicture Failed Error

rameshvishnoi90904, 2016-02-13:
After modifying the below line i get takePicture failed error
Camera camera = RCTCamera.getInstance().acquireCameraInstance(2);

rt2zz, 2016-02-17:
Is this in android? try installing the latest from master `npm install lwansbrough/react-native-camera`

GauravBuche, 2017-02-10:
Check if your app has permissions to access device camera. This applies to new android versions that support setting permissions per app (since Marshmallow, 6.0). The permission for camera could be disabled and should be enabled from the app settings. Settings -> Apps -> [Your App] -> Permissions

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco