react-native-camera/react-native-camera #198: How to show capture button?

juggleru, 2016-02-13:
I'm trying to show capture button like this:

```
render() {
        return (
            <View style={styles.container}>
                <Camera
                    ref={(cam) => {this.camera = cam;}}
                    style={styles.preview}
                    aspect={Camera.constants.Aspect.Fill}
                >
                        <Image
                            source={require('../img/capture.png')}
                            style={{width: 50, height: 50}}
                        />
                </Camera>

            </View>
        );
    }
```
But the image is not visible.

What is wrong?

mikollo, 2016-02-23:
You might try using position='absolute' for this image and make sure that You have proper source.

JBerendes, 2016-02-24:
On Android I also found this to be a problem. Consistency is maybe the more immediate concern.
Absolute positioning is correct but I had to start the component outside the Camera rather than containing it inside. Now that works and looks consistent on iOS and Android for me. I also put all of that inside my own CameraCapture component so I now have something understandable and reusable.

sodik82, 2016-02-25:
@JBerendes thanks - it helped

cubbuk, 2016-03-09:
As @JBerendes mentioned subviews for Android does not work, I had to put my components outside `Camera` component too, but in IOS it works as it is described in the docs. Maybe it is a good idea to update **Subviews** section in docs to clarify this difference.

dcz-switcher, 2016-03-18:
Maybe add another example for android ?

`render() {
        return (
            <View style={styles.container}>
                <Camera
                    ref={cam => this.camera = cam }
                    style={styles.preview}
                    aspect={Camera.constants.Aspect.fill}>
                </Camera>
                <TouchableHighlight 
                    style={styles.actionButton}
                    onPress={this.takePicture.bind(this)}>
                    <Image source={require('./img/camera-icon.png')} />
                </TouchableHighlight>
            </View>
        );
    }`

`const styles = StyleSheet.create({
  container:{
      flex: 1,
  },
  preview: {
    flex: 1,
  },
  actionButton: {
        position: 'absolute',
        bottom: 25,
        padding: 16,
        right: 20,
        left: 20,
        borderRadius: 20,
        alignItems: 'center',
  },
});`

stoffern, 2016-03-23:
This will work:
```
render() {
    return (
		<View style={styles.container}>
			<Camera
			  ref={cam => this._cameraComponent = cam}  {...this.props}
			  style={styles.preview}
			  aspect={Camera.constants.Aspect.fill}>

			</Camera>

		</View>
	
		<Text 
		  style={styles.capture} 
		  onPress={(event) => this._takePicture()}>
		  CAPTURE
		</Text>

    );
}
_takePicture() {
	this._cameraComponent.capture()
	  .then((data) => {
	    console.log(data)
	  })
	  .catch(err => console.error(err));
}
```

clovs, 2017-01-30:
Hello guys,
Did you find a solution?

I want to make an image in front of the camera, to replace this horrible text "CAPTURE"
and also add an image that make a designed frame like in usual camera.

Let me know :-)

dmitry, 2017-01-30:
@clovs use `background-image` in a `Text` element.

clovs, 2017-01-31:
First Thanks Dmitry to reply I'm trying that,
But how can we put others stuff?
Like an image that make a designed frame on the camera or a button to Camera-Roll etc..
[image]

stoffern, 2017-01-31:
@clovs use my method, it works

clovs, 2017-01-31:
yeah, I don't kmow what I was doing, but it's work now.
thanks lot stoffern ;-)

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco

robto09, 2019-09-24:
> yeah, I don't kmow what I was doing, but it's work now.
> thanks lot stoffern ;-)

Hellow! Excuse me, how did you do it?