react-native-camera/react-native-camera #199: onBarCodeRead doesn't work on Andoird

splicefracture, 2016-02-14:
I'm not seeing support for onBarCodeRead on Android.
Are there any plans for this to be to be implemented?

Here is my sample code in case I'm missing something.

```
  render() {
    return (
      <View style={styles.container}>
        <Camera
          ref={(cam) => {
            this.camera = cam;
          }}

          onBarCodeRead={this.onBarCodeRead.bind(this)}
          style={styles.preview}
          aspect={Camera.constants.Aspect.Fill}>
        </Camera>

        <TouchableOpacity onPress={this.takePicture.bind(this)}>   
          <Text>CAPTURE</Text>
        </TouchableOpacity>

      </View>
    );
  } 

  onBarCodeRead(e) {
    console.log(
        "Barcode Found!",
        "Type: " + e.type + "\nData: " + e.data
    );
  }
```

lwansbrough, 2016-02-15:
Feature parity is still a ways off for Android. As far as I know Android doesn't have a barcode scanner built into the default camera implementation, so we'll have to consider how we can make that work too.

rohitgoyal, 2016-02-18:
@lwansbrough 
Google just released Barcode APIs recently. 
https://developers.google.com/android/reference/com/google/android/gms/vision/barcode/Barcode
Check it out.

miletbaker, 2016-02-29:
+1 for integrating Google's vision.barcode API so we can have onBarCodeRead implemented on Android. 

You guys are doing a great job and if you can fit in barcode support on Android that would be amazing. Thanks.

michalkow, 2016-09-29:
+1 for Google Vision . It would be nice to not have to use additional camera module just to scan barcodes on Android.

chirag04, 2016-09-29:
Barcode scan for android already landed on master. If someone can give it a try?

michalkow, 2016-09-29:
I found this topic when checking if the functionality was added, since readme says it is still iOS only.
@chirag04 Thanks for noticing, it does work, I tried it.
I suggest closing this issue.

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco