react-native-camera/react-native-camera #200: RCTCameraManager.m error

gaotinggxl, 2016-02-15:
reject(RCTErrorUnspecified, nil, RCTErrorWithMessage(error.description));

error:   Too many arguments to block call, expected 1, have 3.

lwansbrough, 2016-02-15:
Please specify React version, etc.

gaotinggxl, 2016-02-15:
react version is 0.16

lwansbrough, 2016-02-15:
Ok, sorry about this but we're building master against the latest version of React so we can keep up with important optimizations and API changes in the core. This is an unfortunate side effect of using pre-1.0 software like RN. I'd expect using React Native 0.20 or later will solve your problem, but please understand the camera module is still in a pre-release state as well.

lwansbrough, 2016-02-15:
I'm going to close this as the issue pertains to an earlier version of React Native. Please feel free to re-open the issue if it persists with the current stable version of React.

closed 2016-02-15 by lwansbrough