react-native-camera/react-native-camera #202: iOS - Build fail: too many arguments on block call

andreacab, 2016-02-17:
react-native: 0.14.2
After carefully following the steps mentioned in the readme, I get an error trying to build my project: 
"Too many arguments to block call, expected 1, have 3"

In RCTCameraManager.m there are 8 calls to a reject method with three arguments passed in. Xcode expects only one, any idea?

Thanks for your help.

rt2zz, 2016-02-17:
@andreacab we do not support old versions of react-native, especially right now when this project is a bit in flux. I would recommend upgrading to react-native 0.20 and installing camera from master `npm i lwansbrough/react-native-camera`

andreacab, 2016-02-17:
Thanks so much for your quick answer. Knew that would happen some day. Will update then.

chenbin92, 2016-02-18:
@andreacab @rt2zz  I have the same problem, thank you very much

rt2zz, 2016-02-19:
after reading more it appears the signature of promise reject was changed in 0.19. Your best bet is to either upgrade to rn .19+ or to downgrade react-native camera.

closed 2016-02-19 by rt2zz