react-native-camera/react-native-camera #205: Prompting and forced acceptance of microphone permissions even if captureAudio set to false

abr54, 2016-02-19:
This is related to the change that was made in https://github.com/lwansbrough/react-native-camera/pull/129 - as suggested there I have a use case where I don't need microphone access at all (a barcode scanner).

I've set the ```captureAudio``` prop to false but I'm still being asked to allow microphone access when trying to access the camera. Additionally ```checkDeviceAuthorizationStatus``` returns false in the case where camera permission is given but microphone isn't.

Ideally it'd be possible to just ask and check for camera permission without any reference to the microphone permission.

RGBz, 2016-02-29:
I second this, the API should be updated inline with @lwansbrough's concerns from #129

rt2zz, 2016-03-03:
we can either
a) add an optional options parameter to `checkDeviceAuthorizationStatus`. Something like:`{types: ['audio', 'video']}`
b) split this out into two methods: `requestVideoAccess` `requestAudioAccess`.

Thoughts?

rt2zz, 2016-03-03:
after writing this I think b) is the way to go. It is cleaner and more explicit.

abr54, 2016-03-08:
I think b) sounds reasonable yep.

trabianmatt, 2016-04-04:
This looks like it was covered in 06ad9347a4399745bef0e1d012824e20e5e5a2b2.

trabianmatt, 2016-04-04:
(and 990a1084d5ff5757d3f6267f7ce8d45f7b93b9df)

closed 2016-04-04 by rt2zz

varunsdcet, 2019-02-28:
i am added permission of NSMIcrophone My app is reject . Please help