react-native-camera/react-native-camera #206: RCTCameraModule.stopCapture got 2 arguments, expected 3

bdeezz, 2016-02-19:
React Native version 0.20.0, Android 5.0, Node.js version 5.0.0, react-native-camera (master)

Phoenix1708, 2016-03-07:
Same Issue here React Native Version 0.21.0. Android 6.0.1, Node version 4.3.2, react-native-camera (master)

appbeijing, 2016-04-05:
same with upers

Frosty92, 2016-04-19:
Same issue here, React Native 0.22.2, Android 5.0.1, Node v 4.4.1

cubbuk, 2016-05-24:
Same issue here, React Native 0.26.1, Android 6.0.1. Any idea how to solve it?

zachrnolan, 2016-08-01:
Same issue! 

React Native 0.29.2, Android 4.4.4, Node 5.3.0

Fuco1, 2016-08-07:
Same here

react-native-cli: 1.0.0
react-native: 0.31.0
android 4.4.2
node 5.8.0
on master

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco