react-native-camera/react-native-camera #210: Shutter "effect" executed at the beginning of recording a video

robsquires, 2016-02-23:
### Steps to reproduce
1. Setup camera in video recording mode as per [Usage](https://github.com/lwansbrough/react-native-camera#usage)
2. Start recording, using `camera.capture()`

### Expected behaviour
When I start recording a video, I'd like to apply my own UX to notify the user that recording has started. (show a timer etc). 

### Actual behaviour
When I start recording a video, I get a shutter "effect", similar to that of taking a photo on the native iOS camera. I can of course apply my own additional UX :+1:, but I've no way of disabling this effect

### Environment
- **Node.js version**: 4.2.3
- **React Native version**: 0.19.0
- **React Native platform + platform version**: iOS 9.2.1

### react-native-camera
**Version**: npm version

gimlids, 2016-02-26:
Are you sure this is an "effect"? Or is there something strange happening with the camera's actual exposure? I am seeing what I believe to be a camera exposure issue at the beginning of recording video, resulting in the first frame of every video being essentially black. I'm guessing the iOS camera API is being manipulated in such a way as to cause this.

gimlids, 2016-02-26:
Followup, is this a dup of https://github.com/lwansbrough/react-native-camera/issues/158?

robsquires, 2016-02-27:
@gimlids I've posted the effect here: https://youtu.be/DjZ_Or_zA4c

robsquires, 2016-02-27:
This doesn't happen when you start recording video in the native camera app, so I agree it's likely to be custom manipulation of the API. I've had a look through the RCTCamera and RCTCameraManager but I'm struggling to find too much

robsquires, 2016-02-28:
@gimlids you were right -  #158 solved this :+1:

closed 2016-02-28 by robsquires