react-native-camera/react-native-camera #215: Camera gets slow and eventually gets stuck

rtfpessoa, 2016-02-28:
I am using `react-native-camera`with the helper [react-native-qrcode-reader](https://github.com/lazaronixon/react-native-qrcode-reader) after I take my first shot/qr code scan.
If I go back to the camera view the camera stops working leaving a steady image in the screen.
The rest of the app seems unaffected.

### Steps to reproduce
1. Use `react-native-qrcode-reader` to scan qr code
2. Go back to scan a second qr code

### Expected behaviour
It should behave normally

### Environment
- **Node.js version**: 5.7.0
- **React Native version**: 0.20.0
- **React Native platform + platform version**: iOS 9.3

### react-native-camera
**Version**:  "master"

Am I doing something wrong?

rt2zz, 2016-03-03:
There may be a memory leak (see #216) or there could be an issue with having multiple cameras mounted. Often times the way navigator works you end up with multiple running cameras which is very costly. One solution for this case is to track which route is active and do not render the camera if the route is not active. This is really easy to do with NavigationExperimental, I am less certain about the regular navigator.

rtfpessoa, 2016-03-04:
@rt2zz what do you mean by `NavigationExperimental`?

rt2zz, 2016-03-04:
@rtfpessoa it is not in the docs yet, but the core team is working on a new Navigation library: https://github.com/facebook/react-native/commit/a3085464f6ea36fc6b53cd0c711c048ffb1516f9

agrass, 2016-04-15:
How I can prevent the multiples rendering of cameras?? I'm using react-native-router-flux, but every time I go back to Camera it render again.

agrass, 2016-04-18:
I exposed a already created method to RN to release the camera and prevent having multiple cameras mounted, in some Android devices I have problems because the camera continue working in background when I changed the route.  #264

Lube, 2016-08-31:
Did you eventually went with releaseCamera agrass? Did you find any solution to this problem?

agrass, 2016-09-27:
You are using NavigationExperimental or react-native-router-flux @Lube??  You still having this issue? I'm working with `react-native-router-flux` and to fix it I'm reseting the scene `Action.camera({type: 'reset'})` when I have multiple cameras running.

Lube, 2016-09-27:
Im using the rnrf too, thanks for the tip!
I will try this later this week!

AnisDerbel, 2016-10-29:
is there any solution for that? I'm facing the same problem!

kodatech, 2017-08-03:
Hi, I'm using @agrass solution and seems to work but after some bar codes scanned I started to get a white screen in between I push the button to go to the camera scene and the moment I get the image from the camera, before taking the picture. Looks like after more scanning white screen remains longer. Not sure if it is the same problem but could be. Did anyone face this?

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco

GaneshSaiKumar, 2019-01-12:
happening again
###how?
whenever i go to some other screen and come back to RNCamera page, i'm getting a white screen for 2 seconds and after that it's working fine.

though cam is not being shown, it is still working from 0 seconds but view rendering only after 2 secs.

react-native-camera: 1.8.0
react-navigation:2.6.2

n1ru4l, 2019-01-14:
@GaneshSaiKumar please open a new issue with the template.