react-native-camera/react-native-camera #216: memory leak?

tomprogers, 2016-03-01:
I've got a simple POC app that presents the react-native-camera, along with a hold-to-record button. When I open the app and let it run with the Perf Monitor active, the RAM used by the app slowly climbs over time, even if I don't record anything.

Has anyone else noticed whether this module is leaking memory?

Here's the relevant code:

```
class POC extends Component {
	// invoked on press
	startRecording = () => {
		return this.camera.capture()
		.then((videoUri) => {
			console.log(`capture stored at ${videoUri}`);
		})
		.catch((err) => {
			console.log('it broke!');
			throw Error(err);
		});
	};
	// invoked on release
	stopRecording = () => {
		this.camera.stopCapture()
	};
	
	render() {
		return (
			<View style={Styles.fullscreen}>
				<Camera
					ref={(cam) => { this.camera = cam; }}
					aspect={Camera.constants.Aspect.Fill}
					captureAudio={true}
					captureMode={Camera.constants.CaptureMode.video}
					captureTarget={Camera.constants.CaptureTarget.disk}
					type={Camera.constants.Type.front}
					orientation={Camera.constants.Orientation.auto}
					style={Styles.camera}
					>
					
					<TouchableHighlight onPressIn={this.startRecording} onPressOut={this.stopRecording} style={Styles.record} activeOpacity={0.8} underlayColor="red">
						<Text>RECORD</Text>
					</TouchableHighlight>
				</Camera>
			</View>
		);
	}
	
}
```

### Steps to reproduce
1. Load the app and open the Perf Monitor from the debug menu
2. Note the RAM in used; for me, it starts at ~31MB
3. Don't interact with the app at all, but watch the RAM value.

### Expected behaviour
I'd expect RAM use to remain roughly constant.

### Actual behaviour
RAM in use grows at a steady pace until the OS terminates it.

### Environment
- **Node.js version**: v5.5.0
- **React Native version**: v0.20.0
- **React Native platform + platform version**: iOS 8.2

### react-native-camera
**Version**: master (currently 0.3.8)

corbinschwartz, 2016-03-23:
Our team is able to reproduce by just including the camera package. Still trying to figure out a way to make it not leak.

rpopovici, 2016-03-23:
Guys thanks for finding this. I think I fixed https://github.com/lwansbrough/react-native-camera/pull/233 Just let me know if this fixes the problem

tomprogers, 2016-03-23:
Thanks! Will pull from `eab5ef1` and let you know.

corbinschwartz, 2016-03-24:
@rpopovici, thank you!

rt2zz, 2016-03-24:
merged :+1:

rt2zz, 2016-03-24:
please reopen if it is still an issue

closed 2016-03-24 by rt2zz