react-native-camera/react-native-camera #220: Undefined is not an object(evaluating 'CameraManager.Aspect')

pankajmonga12, 2016-03-04:
i try to make simple react native camera app. when i try to run it gives me tht error.

rt2zz, 2016-03-04:
what version of camera and what version of react-native are you using? I would suggest installing camera from master

pankajmonga12, 2016-03-05:
react-native-cli: 0.1.10
react-native: 0.21.0

i already installed camera from master

lindesvard, 2016-03-07:
Same here. When I import the module the error occurs.

Have react-native 0.21.0 aswell.

Vampirkod, 2016-03-09:
After installing and linking package, important to rebuild native part of app.

freemanon, 2016-03-10:
Same here, just upgraded to 0.21.0

freemanon, 2016-03-10:
Fixed it, had to link the library manually.

moxiaobei2, 2016-05-20:
my react-native is up to 0.26,but has the same error

sofakeer, 2016-06-03:
to fix bug with update all libs to 0.26

lsps9150414, 2016-06-05:
Seeing the same error when run with `react-native run-ios` but not when runing with Xcode.

Hosar, 2016-07-10:
I'ts not working, same error here!.

alistairwoodcock, 2016-07-11:
I was getting this error because I ran `rnpm link` but I managed to fix it by following the manual install instructions but instead making changes to MainActivity.java I made the changes to MainApplication.java and removed 
 `import com.lwansbrough.RCTCamera.RCTCameraPackage;`
`import com.lwansbrough.ReactCamera.ReactCamera;`

from MainActivity.java

This is working so far but I could be doing a bad thing

jqn, 2016-07-12:
Has anyone figured out how to fix this error?

jqn, 2016-07-13:
I was experiencing the same error only in android - the best way to deal with this is to make sure the project is not running when doing the initial install - for ex: stop the android emulator and the android packager running in the terminal.

jqn, 2016-07-13:
In addition to the above the documentation is also inaccurate for react native 0.29.0 - don't make changes to MainActivity.java. Instead make changes to MainApplication.java:
`import com.lwansbrough.RCTCamera.*;`
and

`new RCTCameraPackage()`

dunglt90vt, 2016-07-22:
thanks bro @jqn

prabhaav, 2016-07-31:
Seeing the same issue in IOS. RN 0.3 React 15.2.1. Has anyone found a proper solution for IOS?

gonzaloserrano, 2016-07-31:
Same thing trying to run the iOS version: https://github.com/lwansbrough/react-native-camera/issues/353

gonzaloserrano, 2016-07-31:
@freemanon how did you link it?

jdmarshall, 2016-08-01:
@Vampirkod removing the build directories and rerunning fixed my problems.  Didn't have to hand-edit anything.

yairniz, 2016-08-02:
Fixed here:
1. Check the camera is available (on emulator works fine)
2. Rerun: `react-native run-android`

NatalieFrecka, 2016-09-15:
react-native-cli: 1.0.0
react-native: 0.33.0
react-native-camera: 0.3.8

Happening to me on the iOS simulator. I assume it's because the simulator does not have access to a camera because it works on the iphone itself. Would be nice if it failed gracefully and just showed a black screen or something.

leonacky, 2016-11-17:
same error on IOS
I have install and link the package
After done I have run again the cmd: ract-native run-ios

jjhesk, 2016-11-18:
this is happening again.

leonacky, 2016-11-20:
When I delete node_module folder and re-install, its work fine

annarbt, 2016-11-29:
Deleting android/build and rerunning run-android worked for me

sheerun, 2017-01-23:
It happens for default app with react-native 0.40.0..

taiki-t, 2017-02-21:
https://github.com/lwansbrough/react-native-camera/issues/220#issuecomment-223796532

> Seeing the same error when run with react-native run-ios but not when runing with Xcode.

Same. I solved it by turning `Build Active Architecture Only` option of Pods `Yes` to `No` in Xcode.

`Pods -> Build Settings -> Build Active Architecture Only -> NO`

joshuapinter, 2017-03-01:
If you added React Native to an existing app, as per [Integration With Existing Apps](https://facebook.github.io/react-native/docs/integration-with-existing-apps.html), the automatic installation will not work and you'll need to manually add the project, dependency and package to where ever you have defined your ReactRootView.

I was so used to `rnpm` and `react-native link` being so awesome that this threw me for a loop.

Vivism, 2017-04-28:
Getting this when I simulate a home button press then try to reopen the app on iOS sim.

twiz911, 2017-05-02:
Android manually linking fixed the same error for me.

stevenleeg, 2017-07-18:
Seeing this issue as well on React Native v0.46.4 and React Native Camera v0.9.4. Has anyone found a solution that works?

JamesDorrian, 2017-08-01:
Also have this issue, no solution?

JamesDorrian, 2017-08-02:
Is there any good alternative to react-native-camera that wont derp up my entire project?

capozzid, 2017-08-30:
I have the same issue with RN0.47.2

hayait599, 2017-10-08:
same problem !! :(

monolithed, 2017-10-09:
The same problem with iOS

```
		"react": "^15.3.2",
		"react-native": "^0.37.0",
		"react-native-camera": "^0.4.1",
```

toebbel, 2017-10-09:
same issue with android and
```
  "react-native": "0.48.3"
  "react-native-camera": "10.0.0^",
```

monolithed, 2017-10-10:
@JamesDorrian probably https://github.com/wix/react-native-camera-kit 
 I hate react-native for such errors.

felixsonyusuftosin, 2017-10-24:
Is this problem fixed? because this is not a very stable library ..no practical solution so far

eggybot, 2017-10-31:
this issue still not fix, I need a proper instruction on how to deal wit this issue? thanks

Inlesco, 2017-11-01:
@JamesDorrian `Is there any good alternative to react-native-camera that wont derp up my entire project?`

So far, a good equivalent I found is Expo's RN Camera implementation: https://docs.expo.io/versions/latest/sdk/camera.html

shadowbrush, 2017-11-05:
I encountered this issue on a fresh project. But then I noticed the README suggests to install `0.6` for RN>0.40, but they're currently up to `0.10.0`. With the latest version, the issue vanished. IOW, just `yarn add react-native-camera && react-native link react-native-camera` (and the various permissions, config changes) worked for me on RN 0.50.1.

pavanmehta91, 2017-11-16:
@jqn  You're the savior

badis, 2017-11-16:
I am having the same issue.
I fixed it by running: 
```
react-native link
react-native run-android
```

sebgam, 2017-11-17:
I have the same problem with the emulator in ios but when I run it in a real device it works perfectly

GloomyNAN, 2017-11-22:
I have the same problem too!
```
react-native-cli: 2.0.1
react-native: 0.50.3
```

zenineasa, 2017-12-03:
As @jqn said, in android/src/main/java/com/<appname>/MainApplication.java, add the following:

The import statement at the initial part:
`import com.lwansbrough.RCTCamera.*;`

A new RCTCameraPackage() into the array under getPackages():
`    protected List<ReactPackage> getPackages() {
      return Arrays.<ReactPackage>asList(
          new MainReactPackage(), **new RCTCameraPackage()**
      );
    }`

zenineasa, 2017-12-03:
As @jqn said, in android/src/main/java/com/<appname>/MainApplication.java, add the following:

The import statement at the initial part:

import com.lwansbrough.RCTCamera.*;

A new RCTCameraPackage() into the array under getPackages():

    protected List<ReactPackage> getPackages() {
      return Arrays.<ReactPackage>asList(
          new MainReactPackage(), **new RCTCameraPackage()**
      );
    }

zenineasa, 2017-12-03:
As @jqn said, in android/src/main/java/com/<appname>/MainApplication.java, add the following:

The import statement at the initial part:

    import com.lwansbrough.RCTCamera.*;

A new RCTCameraPackage() into the array under getPackages():

    protected List<ReactPackage> getPackages() {
      return Arrays.<ReactPackage>asList(
          new MainReactPackage(), new RCTCameraPackage()
      );
    }

wilbolabilbo, 2017-12-10:
Installing 0.12.0, following the 'Mostly automatic install with react-native' instructions, I had to run the pod install after linking to get things working on iOS:

```
npm install react-native-camera --save
react-native link react-native-camera
cd ios && pod install
```

sibelius, 2017-12-10:
link properly

closed 2017-12-10 by sibelius

TimDorand, 2018-01-23:
For thoses who still have issues this worked for me: 

1. clean `/android/settings.gradle`
 ```
rootProject.name = 'GrifitupApp'
include ':react-native-camera'
project(':react-native-camera').projectDir = new File(rootProject.projectDir, '../node_modules/react-native-camera/android')

include ':app'
```
to
 ```
rootProject.name = 'GrifitupApp'
include ':app'
```

2. unlink previously linked package: 
```
react-native unlink react-native-camera
```
3. Make sure you don't have any npm dependencies 

Example: 
```
npm WARN react-native-qrcode-scanner@0.0.29 requires a peer of react-native-camera@^0.10.0 but none is installed. You must install peer dependencies yourself.
```
Then: 
```
npm i react-native-camera@^0.10.0
```

4. Follow the README using the react-native link command
```
npm install react-native-camera --save
react-native link react-native-camera
```

xtrycatchx, 2018-04-04:
Did the following, but still encountered the same problem
```
npm install react-native-camera --save
react-native link react-native-camera
```

What worked for me is to manually add **libRNCamera.a** in the **Linked Frameworks and Libraries** of my Target project in xcode.

jaydenmaddison, 2018-04-11:
After linking the libraries, run 'pod install' in the ios directory. This solved it for me.

FlaviooLima, 2018-04-16:
Try doing [this](https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md) 
[In Android it solved my issue]

amerllica, 2018-04-23:
@TimDorand , what is your first step? how I can do it? where I can clean `/android/settings.gradle`

TimDorand, 2018-04-23:
@amerllica Just remove all lines related to `react-native-camera` in the file `/android/settings.gradle`

amerllica, 2018-04-23:
@TimDorand , I think I sucked, I have no `/android/settings.gradle`, even I don't use gradle, I use `expo` for show app inside my app. I don't know why I see this damn error: `Undefined is not an object(evaluating 'CameraManager.Aspect')`

jaydenmaddison, 2018-04-23:
If you are using Expo for your app you need to use the camera component
supplied by them and not the react native one. Here is the link to it
https://docs.expo.io/versions/latest/sdk/camera.

amerllica, 2018-04-23:
@jaydenmaddison , thanks, but how you develop your apps? for example android app! how? my OS is windows 10

[13 more replies]